openwch/arduino_core_ch32 #9: Support for MacOS

IndianTinker, 2023-06-26:
Hey all,
Thanks a lot for the great work porting this amazing chip to Arduino. 
I am unable to use it on my M2 Pro MacBook Pro as when I install the toolchain I get the following error 

```
Downloading packages
Failed to install platform: WCH:ch32v.
Error: 13 INTERNAL: Error downloading tool WCH:riscv-none-embed-gcc@8.2.0: no versions available for the current OS, try contacting tech@wch.cn
```

The [unofficial port](https://github.com/AlexanderMandera/arduino-wch32v003) of the Arduino core works without any problems. 

Can you support MacOS soon? I am building test projects using CH32V003 and want to use more peripherals that supported by Alexander's port.
Thanks!

varna9000, 2023-07-08:
+1 for Mac OS support!

varna9000, 2023-09-20:
Any joy for Mac OS support?

TianpeiLee, 2023-12-13:
> Any joy for Mac OS support?

I encountered a problem while trying to adapt, and I am trying to communicate with MRS

TianpeiLee, 2023-12-18:
Now is ok

closed 2023-12-18 by TianpeiLee