openwch/arduino_core_ch32 #10: No Linux/Ubuntu riscv-none-embed-gcc compiler available

lbattraw, 2023-07-15:
I tried adding this core to my Arduino v1.8.19 board manager on my Ubuntu 22.04/"Jammy Jellyfish" x86 Linux and got the error:
Tool riscv-none-embed-gcc is not available for your operating system.

Is this core only available for Windows?  I already have this version of gcc/binutils installed on my OS, so it's not great being told I can't use the core.  I realize that Arduino ignores any OS-installed compilers, however it's still annoying.  

Is there anything I can do to help get this OS supported for arduino_core_ch32?

taunoe, 2023-07-30:
I have the same problem :(

maxgerhardt, 2023-08-06:
The compiler for Linux and Mac is available, it's just not listed in the package index.

https://raw.githubusercontent.com/openwch/board_manager_files/main/package_ch32v_index.json

The raw toolchains can be gotten from http://www.mounriver.com/download or my mirrors 
* https://github.com/Community-PIO-CH32V/toolchain-riscv-mac
* https://github.com/Community-PIO-CH32V/toolchain-riscv-linux/

Still, the people in this repo need to fix it.

I'm integrating this core into PlatformIO, where a CH32V003 example compiles fine on Windows, Linux and Mac. Currently [work in progress](https://github.com/Community-PIO-CH32V/platform-ch32v/tree/develop/examples/blinky-arduino).

mayermakes, 2023-12-01:
Same behaviour on Arduino IDE 2.2.1
Failed to install platform: 'WCH:ch32v:1.0.3'.
Error: 13 INTERNAL: Error downloading tool WCH:riscv-none-embed-gcc@8.2.0: no versions available for the current OS, try contacting tech@wch.cn

maxgerhardt, 2023-12-01:
You know what, let me just whip up an expanded `package_ch32v_index.json` version that just has links to my repos for the missing toolchains.  That should actually make it go through.

maxgerhardt, 2023-12-01:
Okay.. it just has a problem with the OpenOCD package becuase it does not have a unique root directory.. let me repackage that..

[image]

maxgerhardt, 2023-12-01:
The URL `https://gist.githubusercontent.com/maxgerhardt/e270b2c3294a5f7a6cfd9e2d0d16382e/raw/803a5ca106af2aa3336c508674ad72f0c152ba2b/arduinowch_package_index.json` works for me with Linux.

[image]

[image]

Don't have time to repackage OpenOCD for Mac (and besides the Linux hoster at catbox is just temporary), but it does prove the concept..

TianpeiLee, 2023-12-18:
I'm sorry for waiting until now to address these issues. Both Linux and MAC environments are  available now.

closed 2023-12-18 by TianpeiLee

maxgerhardt, 2023-12-18:
Looks good in regards to toolchain and OpenOCD now, but is it right that all Linux, Windows and MacOS systems reference the `before_install` package with `wch_beforeinstall_linux-1.0.0.tar.gz` which has Linux `.so` library files?

TianpeiLee, 2023-12-18:
I noticed this issue. Other environments try not to include it, but when other environments' IDEs download the installation package based on the same "toolsDependencies", they will report an error. For the sake of simplicity, I have added all of this package.