openwch/arduino_core_ch32 #13: flashing Blink example to CH32V003F4P6-EVT fails

Knochi, 2023-07-28:
Hi,

I tried flashing my CH32V003F4P6-EVT with the blink example.
1. add board URL to Arduino IDE
2. install "CH32 EVT Boards Support" (searching for WCH didn't work)
3. load blink example
4. connect VCC to 3.3V, Gnd to Gnd, PD1 to SWDIO
5. connect WCH-LNK
6. setup board "CH32V00x"
[image]
7. select port, keep anything else as it is
[image]
8. upload

i got this error
[image]

Once I get it running I'm willing to to a "getting started" documentation

tmolteno, 2023-08-01:
What version is the software on the WCH-LinkE. I have found that using it once in the Windows IDE and choosing to update it, fixed problems using openocd. Perhaps this might help?

Knochi, 2023-08-01:
How do I check/update?
I have Linux environment at the moment but a windows pc is available

tmolteno, 2023-08-01:
I am in a Linux environment too (using Debian). I had to borrow a windows machine to do this.

Kind regards
Tim

taunoe, 2023-08-02:
I have the same hardware and have not updated the WCH-LinkE, and it works.
New Arduino IDE 2.*?
 
[image]

maxgerhardt, 2023-08-06:
FYI the WCH-Link firmware updater is contained in the original MounRiver IDE (see www.mounriver.com/download, see https://github.com/Community-PIO-CH32V/platform-ch32v/issues/30)

tmolteno, 2023-08-07:
I ran into this issue after the LinkE had been working for a while. The link-E just stopped working. 

BTW. (separate issue) The firmware updater from the MounRiver IDE on Debian bricked one of my Link-E's which was successfully fixed after I connected it to the Windows IDE.

taunoe, 2023-08-07:
Is the red or blue LED lit? If the blue LED on the WCH-LinkE stays ON that means it is on ARM mode and must be switched to RISC-V mode.

Knochi, 2023-08-07:
Red one

Mit freundlichen Grüßen
David Nottebohm-Knochenhauer

Knochi, 2023-08-20:
Ok. tried flashing the blink example with my windows machine... same result.

So downloaded MounRiver Studio and tried to update the LinkE. Seems the only (easy) way to do so is to create a new project, compile and download.

Then the IDE told me that LinkE is on 2.8 and whether i want to update to 3.0.

FLashing works now but LED didn't blink.

taunoe, 2023-08-20:
PC0 is the built-in LED.

maxint-rd, 2023-10-05:
Tested this yesterday on Windows 7 32-bit and got it working after some obstacles.

First I tried the WCH-Link utility to test the connection to the chip on my SOP20 EVT board (3v3, GND and SWIO to PD1). The LinkE usb device seemed to be recognized (said it was in  RV mode). To query chip info, the link utitlity asked permission to update. The first attempt gave a permanently blue flashing LED on the LinkE. The link utility then would no longer recognize the LinkE. Scary! Then after some restarts and re-inserting usb, the utility asked again. The blue flashing stopped, the utility recognized LinkE and was able to query the chip and read its flash. Hooray!

Then I tried Arduino IDE 1.8.19 with this core. LED_BUILTIN seemed undefined, so I connected some other port to the LED header and tried that one instead. Although the produced binary was quite large, it flashed succesfully and the LED was blinking. Nice!

maxgerhardt, 2023-10-05:
Antidote against big firmwares: Use https://github.com/cnlohr/ch32v003fun instead. (This is a PlatformIO project so you can just clone + open it in VSCode with the PlatformIo extension installed)

maxint-rd, 2023-10-05:
Or use this [alternative CH32V003 Arduino core](https:/github.com/AlexanderMandera/arduino-wch32v003) by Alexander Mandera (which includes [ch32v003fun](https://github.com/cnlohr/ch32v003fun) by CNLohr). Haven't tried it yet, but from what I read the current is still work in progress and at the moment it supports fewer peripherals in an Arduino compatible API.

Personally I would like the core to be fully compatible, so that the huge base of Arduino libraries and examples becomes available. Of course it's very nice that this single core supports different CH32xxx chips. However, the core shouldn't make the binary too large to be useful. See also issue #25.

maxint-rd, 2023-10-11:
FYI: found a fix to reduce binary size. See issue 25 for details.

TianpeiLee, 2024-05-28:
> Ok. tried flashing the blink example with my windows machine... same result.
> 
> So downloaded MounRiver Studio and tried to update the LinkE. Seems the only (easy) way to do so is to create a new project, compile and download.
> 
> Then the IDE told me that LinkE is on 2.8 and whether i want to update to 3.0. ![grafik](https://private-user-images.githubusercontent.com/5979046/261866643-5d2bb61b-a668-4653-98b0-8a713b54a2b4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY4ODg4MDQsIm5iZiI6MTcxNjg4ODUwNCwicGF0aCI6Ii81OTc5MDQ2LzI2MTg2NjY0My01ZDJiYjYxYi1hNjY4LTQ2NTMtOThiMC04YTcxM2I1NGEyYjQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUyOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MjhUMDkyODI0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZjAwYjE5OGQ0NjJlMWM2MzRiNmQwYmM0OWM1NWJiZTYyOWFjZDAyZmE3OWI3YTk3OTdmY2VhZjMwNWFlNWNjMCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Xodr5dBFzIbfxgmdwAHhhf3m9FwoR5dm38odNhdZvMA)
> 
> FLashing works now but LED didn't blink.
> 
> I don't have any documentation from the board, but from looking at the traces it seems that i need to connect one of the LEDs manually to a pin. To what pin LEB_BUILTIN is mappend?
[image]
The LED of this EVT board is not connected to the specified IO and needs to be manually connected.

closed 2024-05-28 by TianpeiLee