openwch/arduino_core_ch32 #15: Parameter Mismatch Issue in attachInterrupt() of WInterrupts.h/cpp

FZDSLR, 2023-08-05:
## Expected Behavior

According to the [documentation](https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/) and [Arduino core api](https://github.com/arduino/ArduinoCore-API/blob/master/api/Interrupts.h) , the function `attachInterrupt()` should accept params like `(pin_size_t interruptNum, voidTemplateFuncPtrParam<T> userFunc, PinStatus mode, T& param)`, and `attachInterrupt(digitalPinToInterrupt(interruptPin), blink, CHANGE)` should be OK.

## Current Behavior
Err occurs when compiling:

```
C:\Users\xxx\AppData\Local\Temp\.arduinoIDE-unsaved202375-6936-ycii6s.ifkd\sketch_aug5a\sketch_aug5a.ino: In function 'void setup()':
C:\Users\xxx\AppData\Local\Temp\.arduinoIDE-unsaved202375-6936-ycii6s.ifkd\sketch_aug5a\sketch_aug5a.ino:15:56: error: cannot convert 'void (*)()' to 'GPIOMode_TypeDef'
   attachInterrupt(digitalPinToInterrupt(interruptPin), blink, CHANGE);
                                                        ^~~~~
In file included from C:\Users\xxx\AppData\Local\Arduino15\packages\WCH\hardware\ch32v\1.0.3\cores\arduino/wiring.h:46,
                 from C:\Users\xxx\AppData\Local\Arduino15\packages\WCH\hardware\ch32v\1.0.3\cores\arduino/Arduino.h:36,
                 from C:\Users\xxx\AppData\Local\Temp\arduino\sketches\0F31CA54BC21916D8E00040838882DBE\sketch\sketch_aug5a.ino.cpp:1:
C:\Users\xxx\AppData\Local\Arduino15\packages\WCH\hardware\ch32v\1.0.3\cores\arduino/WInterrupts.h:27:52: note:   initializing argument 2 of 'void attachInterrupt(uint32_t, GPIOMode_TypeDef, void (*)(), EXTIMode_TypeDef, EXTITrigger_TypeDef)'
 void attachInterrupt(uint32_t pin,GPIOMode_TypeDef io_mode, void (*callback)(void), EXTIMode_TypeDef it_mode, EXTITrigger_TypeDef trigger_mode);
                                   ~~~~~~~~~~~~~~~~~^~~~~~~

exit status 1

Compilation error: cannot convert 'void (*)()' to 'GPIOMode_TypeDef'
```

## Possible Solution

- Changing the form of function parameters？

## Steps to Reproduce

compiling the program:

```
#include <arduino.h>

const byte ledPin = 3;
 
const byte interruptPin = 2;  
 
volatile byte state = LOW;
 
void setup() {
  pinMode(ledPin, OUTPUT);
  pinMode(interruptPin, INPUT_PULLUP); 
  attachInterrupt(digitalPinToInterrupt(interruptPin), blink, CHANGE);
}
 
void loop() {
  digitalWrite(ledPin, state);
}

void blink() {
  state = !state;
}
```

## Context (Environment)

- Platform: WIndows 10
- Arduino ide
  - version：2.1.1
  - date：2023-06-30T16:04:40.277Z
  - CLI  version：0.32.3
- CH32 EVT MCU Boards: version 1.0.3

maxgerhardt, 2023-08-06:
FYI, ArduinoCore-API docs don't apply here at all, the core uses it's own format

https://github.com/openwch/arduino_core_ch32/blob/29c76add9bafaceaa5acda53a53c582947ec3697/cores/arduino/WInterrupts.h#L27-L28

So it'll expect
```cpp
   /* Input + Pull-Up, Interrupt, triggering on Rise or Fall */
  attachInterrupt(interruptPin, GPIO_Mode_IPU, &blink, EXTI_Mode_Interrupt, EXTI_Trigger_Rising_Falling);
```

TianpeiLee, 2024-05-28:
I have reserved interfaces for events or interrupts and triggering modes(rising edge, falling edge, or both), This is more flexible, and in fact, it does lose compatibility.

closed 2024-05-28 by TianpeiLee

jgromes, 2024-11-07:
I disagree with the statement that "ArduinoCore-API docs don't apply here at all". Is this an Arduino core or not? If it is, it needs to follow the Arduino API - that's what makes it an Arduino core. If you pick and choose some part of it, while ignoring others, it is not compatible and will just lead to people opening issues in this project thinking something is broken.

Is there a specific reason the expected `attachInterrupt` signature cannot be used to make this compatible? Or at least an override/version with default parameters?

maxgerhardt, 2024-11-07:
It says Arduino core on the packaging, but it does not adhere to the API definitions laid out in the ArduinoCore-API repo in this case. Other APIs are matching more closely.

I agree 100% that this core should follow the regular API definition. No sane Arduino sketch writes `  attachInterrupt(interruptPin, GPIO_Mode_IPU, &blink, EXTI_Mode_Interrupt, EXTI_Trigger_Rising_Falling);`.

maxint-rd, 2025-05-06:
Can this issue be reopened with proposal to have an Arduino compatible override for attachInterrupt with default parameters as @jgromes [suggested](https://github.com/openwch/arduino_core_ch32/issues/15#issuecomment-2463158188)?

AlexPar, 2025-05-30:
> Excellent! Perhaps this can be suggested in the form of a PR to facilitate easy integration by [@yangliangquan](https://github.com/yangliangquan) ?

I don't mind, but I'm not good at GitHub.

maxint-rd, 2025-05-30:
Okay. I'm currently not available to test this, but as soon as I find some time in the coming weeks, I want to set up an environment to do some testing. I can then test this on V002, V003, X033, V006 and perhaps V203. If no one else has done it by then, I'll submit the PR. Thank you for your contribution to improve this core!