openwch/arduino_core_ch32 #21: yield()没有实现

juxiangwu, 2023-08-27:
yield()函数没有实现，什么时候可以实现？

kerryland, 2023-12-27:
I'd like to see `yield` implemented too. I suspect it can just be a no-op.

maxint-rd, 2024-01-09:
Upvote for yield(). It would be very nice if this Arduino core would be fully compatible with the other Arduino cores, so that sketches can be compiled without errors. 

In the regular Arduino core yield() is called by the delay() function in wiring.c. In Arduino.h it is implemented as a void function that does nothing, When using the ESP core, calling yield() is required in lengthy loops to do network tasks and keep the watchdog happy. 

I suggest that in this core it is at least implemented as an empty function. Alternatively it could be part of an implementation of Arduino friendly watchdog functionality and let it feed the dog by calling WWDG_Feed();...

TianpeiLee, 2024-03-04:
i have add yield(default weak) function into core library

closed 2024-03-04 by TianpeiLee