openwch/arduino_core_ch32 #22: Guidance for new variant

theaudiobible, 2023-08-28:
Thanks for the useful Arduino core that you have provided.  I am trying to add a variant for CH32V203RB and have made a PR https://github.com/openwch/arduino_core_ch32/pull/20 and would appreciate some guidance on how to make variant_CH32V203RB.h correctly.

kklam4697, 2024-03-03:
@TianpeiLee hi hi, Can you please help to provide a guidence for making new variant file? I also want to make one for ch32x033. Or, maybe you can help to make variant file for us? Thanks a lot

theaudiobible, 2024-03-04:
@kklam4697, @TianpeiLee and @maxgerhardt, I think that I managed to successfully make the variant for CH32V203RB here https://github.com/theaudiobible/arduino_core_ch32/tree/CH32V203RB.  I have been using it along with https://github.com/theaudiobible/platform-ch32v/tree/CH32V203RBT6.

I am happy to work with you to to make any necessary corrections merge these changes back into the original source trees.  Let me know how you would like to proceed.

TianpeiLee, 2024-03-04:
Thank you for your contribution. Due to work reasons, I may not be able to handle it in a timely manner. Sorry!
I have checked  CH32V203RB and it's OK, Regarding the pin issue, I defined it through the development board. If it is a WCH black development board, it usually has Arduino compatible IO ports on it, and this type of board selects these Arduino compatible ports.

theaudiobible, 2024-03-04:
Thanks for merging the CH32V203RB branch, @TianpeiLee.

closed 2024-05-29 by TianpeiLee

maxint-rd, 2024-10-11:
> [kklam4697](/kklam4697) ... I also want to make one for ch32x033. Or, maybe you can help to make variant file for us? Thanks a lot

@kklam4697 - How far did you get? I recently got some CH32X033F8P6 TSSOP20 chips and noticed that the X035 board definitions cannot be used properly. Too many pins just don't work. So I'm now making a variant for the X033 F8P6 which eventually I hope to publish as a PR. I'm starting with digital I/O and then working to see which pins will do ADC and PWM (analogRead() and analogWrite()). Other peripherals will have to wait for now.

rambros3d, 2025-05-23:
Hello, I am trying to add the CH32X035**C8T6** variant.
[image]

There are 26 pin numbers in the CH32X035**G8** variant files
```
/* CH32VX035G8 Pins */
#define PA0                     PIN_A0
#define PA1                     PIN_A1
#define PA2                     PIN_A2
#define PA3                     PIN_A3
#define PA4                     PIN_A4
#define PA5                     PIN_A5
#define PA6                     PIN_A6
#define PA7                     PIN_A7
#define PB0                     PIN_A8
#define PB1                     PIN_A9
#define PB3                     10
#define PB4                     11
#define PB6                     12
#define PB7                     13
#define PB8                     14
#define PB9                     15
#define PB10                    16
#define PB11                    17
#define PB12                    18
#define PC0                     PIN_A10
#define PC3                     PIN_A13
#define PC14                    21
#define PC15                    22
#define PC16                    23
#define PC17                    24
#define PC18                    25
#define PC19                    26
```

The **C8T6** variant has some extra pins compared to the G8 variant.
PA0-PA23
PB0-PB13
PC6, PC7, PC15-PC19

Can someone help me with this variant. How do I number the extra pins?

rambros3d, 2025-05-25:
I also had a doubt why there were two similar names for the same pin (PA1, PA_1), after looking through the files I understood. But I had doubts about the pin numbering.

> For your chip variant I would probably try to follow the current X035G8 definition and add the additional pins with subsequent numbering.

I will add the extra pins subsequently to the X035G8 definition.

> BTW. To request more help you might want to add a new issue. Closed issues may get less response.

Right I will do that. I have a basic working board definition now.

@maxint-rd  Thanks for the detailed explanations.