openwch/arduino_core_ch32 #30: Servo.h library not supports. HELP PLS !

luutrihieu, 2023-10-28:
WARNING: library Servo claims to run on avr, megaavr, sam, samd, nrf52, stm32f4, mbed, mbed_nano, mbed_portenta, mbed_rp2040, renesas, renesas_portenta, renesas_uno architecture(s) and may be incompatible with your current board which runs on ch32v architecture(s).
In file included from C:\Users\luutr\AppData\Local\Temp\.arduinoIDE-unsaved2023929-9732-14s9i1j.0ez7\Sweep\Sweep.ino:10:
C:\Users\luutr\AppData\Local\Arduino15\libraries\Servo\src/Servo.h:79:2: error: #error "This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor."
 #error "This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor."
  ^~~~~

exit status 1

Compilation error: exit status 1

0x0fe, 2023-12-12:
well, that has nothing to do with CH32 core, what the library warns you about is that it has not been written for this architecture and thus will not work. This "servo library" is very dependant on hardware timers/PWM channels of specific targets, and you cannot expect WCH to do anythignaboutthis external library, if you want support for CH32 MCUs in it and you are not able to do the porting yourself your best bet is to open a ticket in this library issue tracker and request for that (i doubt anyone will btoher, but you can try). 
Otherwise, the best course of action is to do it yourself. THe CH32 SDK provides multiple examples on how to use timers and PWM, they are easy to use. You can just check them and start from that.

TianpeiLee, 2024-06-01:
Maybe this library will be provided in the future

closed 2024-06-01 by TianpeiLee