openwch/arduino_core_ch32 #33: IS there a plan for for firmware upload via CDC for the MCU supporting USB?

0x0fe, 2023-11-27:
SO, in arduino the "traditional" firmware upload method is via UART, be it standard UART or CDC when the MCU provides native USB. Upload via USB is supported by the rom bootloader on older WCH MCUs uch as CH5xx, i assume it is also supported on newer MCUs supporting USB such as CH32X035. 
To be convenient to use tact switches are not really an option, so reset to upload mode has to be set via a CDC command, usually this is done by setting the CDC baudrate to an odd / unused value such as 4800, in the handler we just set an RTC flag or similar and reset, so that after the reset the bootloader stays in upload mode , waiting for the datas. 
I have done that some years ago on various STM32 for a client project and it was not particularly hard to implement, is there a plan to add this mechanism to the WCH core? It would be valuable.

maxint-rd, 2023-12-05:
For what it's worth, I really appreciate the effort WCH made to provide an Arduino core. The CH32v003 and it's family is a nice addition to the Arduino environment and delivers great value for money. I encourage WCH to dedicate some more attention to making this core more Arduino compatible, by implementing more of the features Arduino users are familiar with. The current list of issues shows that users are willing to help with improvements, but it seems that core developers have insufficient time available to respond. 

Adding support for USB would be nice. Charles Lohr has shown that it certainly is possible:
   https://github.com/cnlohr/rv003usb
Next to a compact bootloader for uploading code, he also demonstrates the use as HID device such as a gamepad.

0x0fe, 2023-12-05:
@maxint-rd  
The Lohr implementation is bitbanged and cannot serve as a basis forCH32X035 and other MCU featuring an USB PHY. and i highly doubt WCH will bother to implement bit banged USB stack on CH32V003 serie.
Lohr implementation could be used as is on CH32V003, but the USB stack has to be added to the core to perform the reset from CDC.

maxint-rd, 2023-12-05:
Okay, Thank you for this informative reply. Good to know.
(I really hope WCH keeps supporting this Arduino core and give developers a bit more time to participate in discussions)

TianpeiLee, 2024-06-01:
All chips with a bootloader support serial port and USB downloads, but for CH32V103, etc., the level of the boot pin needs to be switched to remap the startup address of the program, which requires auxiliary hardware on the board. The great thing about using WCH-LINKE is that you only need a few wires to achieve these functions, and it provides a serial port

closed 2024-06-01 by TianpeiLee

TianpeiLee, 2024-06-06:
For example, the boot mapping of the STM32 series of microcontrollers we are familiar with is also controlled through pins, which is relatively common among MCUs.
Usually mass production burning tools are used to perform programming operations in mass production. User program upgrades may be through IAP, because each solution provider may have its own firmware upgrade platform, or upgrade verification process and protocol.

jobitjoseph, 2025-09-04:
> So, in arduino the "traditional" firmware upload method is via UART / CDC (when the MCU provides native USB).
> 
> On older WCH MCUs, such as CH5xx, the rom bootloader does support this, i assume it is also supported on newer MCUs such as CH32X035. To be convenient upload should be automatic (no user action on tact switch). If done via UART, no problem, we can use the control lines of the USB-UART bridge to toggle reset and BOOT, on CDC it should be done the baudrate change cdc command, using an odd/unused value such as 4800. When the host sets baudrate to 4800 the USB stack writes a persistant flag (usually in RTC memory) and resets the MCU, after reset the bootloader see the flag and stay in upload mode rather than jumping to the user application, waiting for datas.
> 
> I have done that some years ago on various STM32 for a client project and it was not particularly hard to implement, is there a plan to add this mechanism to the WCH core? It would be valuable.

Here is the new CDC library for CH32X035. You can either manually install it or install it through the Arduino library manager. It uses the native USB PHY for CDC serial device.
https://github.com/jobitjoseph/CH32X035_USBSerial