openwch/arduino_core_ch32 #34: instruction on basic upload and serial support?

linhz0hz, 2023-11-30:
It seems that I have some trouble getting the most basic serial print "hello world" to work.
1) The upload does not work out of the box. I have to use WCH-Link utility to upload manually.
2) After the upload, the simplest Serial print also does not seem to work.

I have programmed CH32V307 using MounRiver studio quite a bit before and I had no trouble, It would be great to have some instruction to walk through the basics.

TianpeiLee, 2023-12-07:
> It seems that I have some trouble getting the most basic serial print "hello world" to work.
>
> 1. The upload does not work out of the box. I have to use WCH-Link utility to upload manually.
> 2. After [...]

may be you can refer to this vedio

TianpeiLee, 2023-12-07:
> It seems that I have some trouble getting the most basic serial print "hello world" to work.
>
> 1. The upload does not work out of the box. I have to use WCH-Link utility to upload manually.
> 2. After [...]

may be you can refer to this vedio
https://www.youtube.com/watch?v=IVlvWfM-m0w&t=5s

Khalinda, 2024-02-26:
Maybe this will help someone else.  I am using a CH32V307 board on Windows 11.  A simple "Hello World" sketch was not printing anything on the Arduino Serial Monitor.  Adding a delay before the first print seems to fix the problem.  

```
void setup()
{
	Serial.begin(115200);
	delay(1000);
	Serial.println("Hello World!");
}

void loop()
{
}
```

TianpeiLee, 2024-06-01:
thanks all!

closed 2024-06-01 by TianpeiLee