openwch/arduino_core_ch32 #36: I2C error twi.c:(.text.i2c_custom_init+0xc): undefined reference to `PinMap_I2C_SDA'

0x0fe, 2023-12-05:
So, we use CH32X035 and there is an error when intilizing the I2C on the ports we use.
What is the problem?

```
#include <Wire.h>

#define SDA         PC16
#define SCL         PC19

TwoWire i2c = TwoWire(SDA,SCL);

void setup() {

  Serial.setTx(TX1);
  Serial.setRx(-1);
  Serial.begin(115200);
  Serial.println("system init");
  Serial.println(DBGMCU_GetDEVID());
  i2c.begin();
}

void loop() {
}

```

0x0fe, 2023-12-05:
so, i found the problem, for some reason the I2C was disabled in the WCH HAL for CH32X035, may i know why?

0x0fe, 2023-12-07:
mmh, are the CH32VX035 currently in production version A?

TianpeiLee, 2023-12-08:
> mmh, are the CH32VX035 currently in production version A?

No, as far as I know, the sales and sample distribution of version A are currently restricted

0x0fe, 2023-12-08:
@TianpeiLee  ok thanks for the clarification