openwch/arduino_core_ch32 #37: Does the CH32V203 USB support CDC profiles, I want to emulate as a serial port.

hacktronics, 2023-12-10:
Does the CH32V203 USB support CDC profiles, I want to emulate as a serial port. Is there any example for the same. How can I connect to my computer the CH32V203 and exchange sensor readings.

0x0fe, 2023-12-11:
you havea CDC device profile example in the sdk, obviously not in this arduino core. You can run the SDK example in an arduino project.

hacktronics, 2023-12-12:
Sorry for this but did you meant this: https://github.com/openwch/ch32v20x/blob/main/EVT/EXAM/USB/USBD/SimulateCDC/User/UART/UART.c
How to run this SDK example in Arduino.

0x0fe, 2023-12-12:
yes sure, well to run it in arduino you just have to integrate this into your project. Or make a library with it? I would not advise to try porting this to CPP, there is zero advantage and a large code ize penalty. Keep it in C and declare the calls you need to make from CPP as extern C

hacktronics, 2023-12-16:
tried copying the USBLib to Arduino, but unfortunately its doesn't work and there are lot of things, which are beyond my understanding.

hacktronics, 2023-12-16:
In mounriver, I was able to emulate a serial port, but didn't understand what the program does. THere is no docs or comments. Is there any sample that just create a serial port, and have interrupt on data receive, even in core sdk.

TianpeiLee, 2024-06-01:
I will consider adapting the USB characteristics of the chip later. If I have enough time, I plan to abstract a USB layer to be compatible with all chips or provide library files for independent chips.

closed 2024-06-01 by TianpeiLee

ralphrmartin, 2025-01-04:
Why is this closed? There is not yet any USB layer or CDC for these chips in Arduino as far as I can see.