openwch/arduino_core_ch32 #40: CH32V003 setting I2C to SCL:PC2 SDA:PC6 leads to error "ERROR: at least one I2C pin has no peripheral"

0x0fe, 2023-12-12:
So on CH32V003, i set I2C to 

```
#define SCL     PC2
#define SDA     PC6
```

And i get the error 

`ERROR: at least one I2C pin has no peripheral`

I tried to set it by 3 differnt way, as argument of the contructor
TwoWire i2c = TwoWire(PC6,PC2); //sda scl
as argument of the init function

```
TwoWire i2c = TwoWire();

void setup() {
  i2c.begin(SDA,SCL);
```
and as default pins in variant_CH32V003F4.h

```
#ifndef PIN_WIRE_SDA
  #define PIN_WIRE_SDA          PC6 
#endif
#ifndef PIN_WIRE_SCL
  #define PIN_WIRE_SCL          PC2
#endif
```
But i still get this error. 
As far as i know the cause would be that PC6 cannot be mapped to SDA, however, when i check the alternatives functions of PC6 i can clearly see SDA is available  so what is the problem?

kklam4697, 2024-02-19:
My understanding...
You are now want yo use PC6 as SDA, however, to use it, require Pinremap...
```c
void GPIO_PinRemapConfig(uint32_t GPIO_Remap, FunctionalState NewState)
```
This function should be call during GPIO init. Witch in this WCH arduino core, it will be call in cores\arduino\ch32\pinmap.c
```c
void pin_function(PinName pin, int function)
```
But the question is, how the core knows if this pin you want to remap is a vaid SDA pin? 
The core had pinMap in variants\CH32V00x\CH32V003F4\PeripheralPins.c
```c
WEAK const PinMap PinMap_I2C_SDA[] = {
  {PC_1, I2C1, CH_PIN_DATA(CH_MODE_OUTPUT_50MHz, CH_CNF_OUTPUT_AFOD, NOPULL, AFIO_NONE)},
  {NC,   NP,   0}
};
```
As you can see SDA only accept PC1 as SDA, same as SCL it only accept PC2 as SCL. 

I have a guess, If we add code as below to the call. It can supput PC6 remap to SDA as you expected. 
```c
WEAK const PinMap PinMap_I2C_SDA[] = {
  {PC_1, I2C1, CH_PIN_DATA(CH_MODE_OUTPUT_50MHz, CH_CNF_OUTPUT_AFOD, NOPULL, AFIO_NONE)},
  {PC_6, I2C1, CH_PIN_DATA(CH_MODE_OUTPUT_50MHz, CH_CNF_OUTPUT_AFOD, NOPULL, AFIO_NONE)},
  {NC,   NP,   0}
};
```
Hope this can help...

0x0fe, 2024-02-19:
well, yes that is how to add more AF in the arduino core, however in the underlaying SDK there is a bit to set, to select which AF to use for a given peripheral, and i dont see how the core would do that here.

TianpeiLee, 2024-05-31:
When initializing the I2C function, the function will query PinMap_I2C_SDA[] and PinMap_I2C_SCL[] based on pin_name. If the corresponding pin is not found in the array, an error will be reported. At this point, a new mapping relationship needs to be added to the array. For example, if you expect SDA to use PC6, then your PinMap_I2C_SDA[] should be like this:

WEAK const PinMap PinMap_I2C_SDA[] = {
  {PC_1, I2C1, CH_PIN_DATA(CH_MODE_OUTPUT_50MHz, CH_CNF_OUTPUT_AFOD, NOPULL, AFIO_NONE)},
  {PC_6, I2C1, CH_PIN_DATA(CH_MODE_OUTPUT_50MHz, CH_CNF_OUTPUT_AFOD, NOPULL, AFIO_FullRemap_I2C1_ENABLE)},
  {NC, NP, 0,0}
};

TianpeiLee, 2024-05-31:
BTW, when adding pin mapping yourself, you need to refer to the corresponding manual and PinAF_CH32V00x.h
[image]
[image]

closed 2024-05-31 by TianpeiLee

TianpeiLee, 2024-05-31:
And it is not an arbitrary mapping of one of the pins, it is a group mapping. For example, if you want to use PC6 as SDA, then you have to use PC5 as SCL.

0x0fe, 2024-06-05:
@TianpeiLee 
yes, that is what i did, 3 month ago.