openwch/arduino_core_ch32 #41: How to use other than TIM1_CH1 for PWM?

hacktronics, 2023-12-16:
On CH32V203, I am trying to use pin A6 for pwm, need 4 pwm channels, so far has been able to use standard code on pin A8, tim1_CH1 only. How to have more PWM outputs.
I am trying the following code, but it does not work.
`void TIM2_PWMOut_Init(u16 arr, u16 psc, u16 ccp)
{
  GPIO_InitTypeDef GPIO_InitStructure={0};
  TIM_OCInitTypeDef TIM_OCInitStructure={0};
  TIM_TimeBaseInitTypeDef TIM_TimeBaseInitStructure={0};

  RCC_APB2PeriphClockCmd( RCC_APB2Periph_GPIOA | RCC_APB1Periph_TIM3, ENABLE );

  GPIO_InitStructure.GPIO_Pin = GPIO_Pin_6;
  GPIO_InitStructure.GPIO_Mode = GPIO_Mode_AF_PP;
  GPIO_InitStructure.GPIO_Speed = GPIO_Speed_50MHz;
  GPIO_Init( GPIOA, &GPIO_InitStructure );

  TIM_TimeBaseInitStructure.TIM_Period = arr;
  TIM_TimeBaseInitStructure.TIM_Prescaler = psc;
  TIM_TimeBaseInitStructure.TIM_ClockDivision = TIM_CKD_DIV1;
  TIM_TimeBaseInitStructure.TIM_CounterMode = TIM_CounterMode_Up;
  TIM_TimeBaseInit( TIM3, &TIM_TimeBaseInitStructure);

#if (PWM_MODE == PWM_MODE1)
  TIM_OCInitStructure.TIM_OCMode = TIM_OCMode_PWM1;

#elif (PWM_MODE == PWM_MODE2)
  TIM_OCInitStructure.TIM_OCMode = TIM_OCMode_PWM2;

#endif

  TIM_OCInitStructure.TIM_OutputState = TIM_OutputState_Enable;
  TIM_OCInitStructure.TIM_Pulse = ccp;
  TIM_OCInitStructure.TIM_OCPolarity = TIM_OCPolarity_High;
  TIM_OC1Init( TIM3, &TIM_OCInitStructure );

  TIM_CtrlPWMOutputs(TIM3, ENABLE );
  TIM_OC1PreloadConfig( TIM3, TIM_OCPreload_Disable );
  TIM_ARRPreloadConfig( TIM3, ENABLE );
  TIM_Cmd( TIM3, ENABLE );
}`

sashahn, 2024-04-02:
Hi, same issue here. Did you find solution?
Thanks

TianpeiLee, 2024-06-01:
[image]

Hi, @hacktronics I think your program clock enable should be enabled like this:
```
  RCC_APB2PeriphClockCmd( RCC_APB2Periph_GPIOA , ENABLE );
  RCC_APB1PeriphClockCmd( RCC_APB1Periph_TIM3,ENABLE);
```
Because the clock of TIM3 is divided by APB1_CLK.

closed 2024-06-01 by TianpeiLee