openwch/arduino_core_ch32 #44: Installing, or updating, to version 1.0.4 fails under Windows (11)

pkootstra, 2024-01-04:
When installing, or updating to, version 1.0.4, the installation fails when using Windows (11), with the following console output:

`Tool WCH:riscv-none-embed-gcc@8.2.0 already installed
Tool WCH:openocd@1.0.0 already installed
Downloading packages
WCH:beforeinstall@1.0.0
WCH:ch32v@1.0.4
Installing WCH:beforeinstall@1.0.0
Failed to install platform: 'WCH:ch32v:1.0.4'.
Error: 13 INTERNAL: Cannot install tool WCH:beforeinstall@1.0.0: extracting archive: Create link C:\Users\UserName\AppData\Local\Arduino15\tmp\package-332495206\wch_beforeinstall_linux-1.0.0\libhidapi-hidraw.so: symlink libhidapi-hidraw.so.0.0.0 C:\Users\UserName\AppData\Local\Arduino15\tmp\package-332495206\wch_beforeinstall_linux-1.0.0\libhidapi-hidraw.so: A required privilege is not held by the client.
`

Apparently the installer attempts to install linux binaries in windows...

dquadros, 2024-01-05:
Same here on a "fresh" install. Selecting version 1.0.3 after the bad install will complete ok.

Khalinda, 2024-02-21:
It looks like Tianpei made a misteak [sic].  I Googled "beforeinstall", and the explanation on this page:

https://github.com/openwch/arduino_core_ch32

implies that the beforeinstall script is only needed on Linux.  I tracked down the script on my PC in this directory:  

`...\AppData\Local\Arduino15\packages\WCH\tools\beforeinstall\1.0.0\`

I edited the file "Start.sh" and I commented out all of the code in the file (prefixing each line with "#").  This next step may not be necessary, but I deleted the "ch32v" directory found here:

`...\AppData\Local\Arduino15\packages\WCH\hardware\`

Finally, I reinstalled CH32 (version 1.0.4) using the Arduino board manager and it installed without a problem.  It seems to be working.  At least I was able to compile a sketch and load it to a CH32 board.  
`

DeqingSun, 2024-02-23:
The install seems OK with "Run as Administrator". On my windows machine the Administrator is another user so the package is installed somewhere else and I need to copy the package in ```C:\Users\{Your USER}\AppData\Local\Arduino15\packages``` to the right location manually.

Ivoz, 2024-04-05:
Right, so it needs to skip trying to symlink linux .so files from `wch_beforeinstall_linux-1.0.0.tar.gz` when run on Windows...

TianpeiLee, 2024-05-31:
@Ivoz This is not provided for Windows, but at the time I remembered that if I did not add this link in the Windows resources, the arduino ide seemed not to recognize the json file normally. Maybe the method I tested at the time was wrong. I will confirm this again.(#89 Same problem)

suprimex, 2024-06-01:
Why it was closed?  The problem is still actual!
Fresh install:

Downloading packages
WCH:beforeinstall@1.0.0
WCH:ch32v@1.0.4
Installing WCH:beforeinstall@1.0.0
Failed to install platform: 'WCH:ch32v:1.0.4'.
Error: 13 INTERNAL: Cannot install tool WCH:beforeinstall@1.0.0: extracting archive: Create link C:\Users\kchistya\AppData\Local\Arduino15\tmp\package-3016519650\wch_beforeinstall_linux-1.0.0\libhidapi-hidraw.so: symlink libhidapi-hidraw.so.0.0.0 C:\Users\kchistya\AppData\Local\Arduino15\tmp\package-3016519650\wch_beforeinstall_linux-1.0.0\libhidapi-hidraw.so: A required privilege is not held by the client.

TianpeiLee, 2024-06-03:
If the IDE requires the library structure to be consistent in each environment, then there is nothing I can do. Maybe I can only install it in administrator mode.

DeqingSun, 2024-06-03:
> If the IDE requires the library structure to be consistent in each environment, then there is nothing I can do. Maybe I can only install it in administrator mode.

They why the link for the 3 dependencies of beforeinstall should be the same link? Here is how I change the link to an empty archive and it installs fine with a local server running on port 9000.

```
{
          "name": "beforeinstall",
          "version": "1.0.0",
          "systems":
          [
            {
              "host": "x86_64-linux-gnu",
              "url": "https://codeload.github.com/TianpeiLee/wch_beforeinstall_linux/tar.gz/refs/tags/1.0.0",
              "archiveFileName": "wch_beforeinstall_linux-1.0.0.tar.gz",
              "checksum": "SHA-256:17EF3A94BC46B9EB9AB1FDECC48FCD3880EEB8FCC36E6648626320F0BF2864DC",
              "size": "507269"
            },
            {
              "host": "i686-mingw32",
              "url": "http://localhost:9000/wch_beforeinstall_empty.tar.gz",
              "archiveFileName": "wch_beforeinstall_empty.tar.gz",
              "checksum": "SHA-256:87f2e600b449220e88b20c5c867eb879b4acad177824fab65a9b3713d53246fd",
              "size": "186"
            },
            {
              "host": "x86_64-apple-darwin",
              "url": "http://localhost:9000/wch_beforeinstall_empty.tar.gz",
              "archiveFileName": "wch_beforeinstall_empty.tar.gz",
              "checksum": "SHA-256:87f2e600b449220e88b20c5c867eb879b4acad177824fab65a9b3713d53246fd",
              "size": "186"
            }				
          ]
        }		
```
[package_ch32v_index.json](https://github.com/user-attachments/files/15534855/package_ch32v_index.json)
[wch_beforeinstall_empty.tar.gz](https://github.com/user-attachments/files/15534856/wch_beforeinstall_empty.tar.gz)

TianpeiLee, 2024-06-04:
I know, but this essentially makes it download a file, and I was expecting it not to download unnecessary files.

DeqingSun, 2024-06-04:
Yes I tried to remove it completely for Windows and Mac, but Arduino throw an error. So I used an empty file.
I think an empty file is better than an incorrect file. 
Or this issue should be submitted to Arduino support.

TianpeiLee, 2024-06-04:
It seems that my previous memory was correct. 
Yes, an empty file is better than an incorrect file. When the tool chain is updated, I will change it to an empty file.

fcassia2017, 2024-07-01:
Confirming it happens on Windows 11. Fresh install of Arduino IDE.
I'm trying to bring newbies into the CH32v003 platform and this is a show stopper.
//
Installing WCH:openocd@1.0.0
Configuring tool.
WCH:openocd@1.0.0 installed
Installing WCH:beforeinstall@1.0.0
Failed to install platform: 'WCH:ch32v:1.0.4'.
Error: 13 INTERNAL: Cannot install tool WCH:beforeinstall@1.0.0: extracting archive: Create link C:\Users\Nando\AppData\Local\Arduino15\tmp\package-3340045858\wch_beforeinstall_linux-1.0.0\libhidapi-hidraw.so: symlink libhidapi-hidraw.so.0.0.0 C:\Users\Nando\AppData\Local\Arduino15\tmp\package-3340045858\wch_beforeinstall_linux-1.0.0\libhidapi-hidraw.so: A required privilege is not held by the client.
//

jimcheesehead, 2025-03-19:
Clean install on two Windows machines as non-privileged user produces the following error
Error: 13 INTERNAL: Cannot install tool WCH:beforeinstall@1.0.0: extracting archive: Create link C:\Users\jimgregor\AppData\Local\Arduino15\tmp\package-1148299858\wch_beforeinstall_linux-1.0.0\libhidapi-hidraw.so: symlink libhidapi-hidraw.so.0.0.0 C:\Users\jimgregor\AppData\Local\Arduino15\tmp\package-1148299858\wch_beforeinstall_linux-1.0.0\libhidapi-hidraw.so: A required privilege is not held by the client.

Same error as Administrator

This is a show stopper!

ednieuw, 2025-03-26:
Same here with version 1.0.4
Installing 1.0.3 solves the problem but updating it to 1.0.4 still gives an error

Error: 13 INTERNAL: Cannot install tool WCH:beforeinstall@1.0.0: extracting archive: Create link C:\Users\xxxxx\AppData\Local\Arduino15\tmp\package-3944909970\wch_beforeinstall_linux-1.0.0\libhidapi-hidraw.so: symlink libhidapi-hidraw.so.0.0.0 C:\Users\xxxxx\AppData\Local\Arduino15\tmp\package-3944909970\wch_beforeinstall_linux-1.0.0\libhidapi-hidraw.so: A required privilege is not held by the client

maxint-rd, 2025-04-01:
Hello @ednieuw - unfortunately I don't have a fresh system to reproduce this issue. 
However, in [this post](https://github.com/openwch/arduino_core_ch32/issues/110#issuecomment-2768025106) you can read about someone needing to delete some folder deep down the file system, in  order to install the latest code. Could that be helpful to this issue too? The folder seems to be some sort of cache that isn't automaticaly removed when restarting the IDE. (Also make sure to run  as Adminitrator)

ednieuw, 2025-04-01:
Thanks @maxint-rd

redo33, 2025-04-12:
The package from UIAPduino can be installed successfully on Windows.

https://www.uiap.jp/en/uiapduino/pro-micro/ch32v003/v1dot4

However, it requires manual firmware upload using the WCH-Link Utility.

FYI.

closed 2025-05-14 by yangliangquan

DeqingSun, 2025-05-16:
@yangliangquan This is an installation error due to wrong beforeinstall packages for windows and mac. This is not a duplicate of PR #87 , which is a upload error.

maxint-rd, 2025-07-14:
Recently I performed a fresh installation of 1.0.4 on a Windows 11 VM. I don't recall such message. Important thing is to **run the IDE as Administrator** when installing it.

--------

For the support of V002/V006 in PR #200, I wrote in [the readme](https://github.com/maxint-rd/arduino_core_ch32/edit/support-CH32V002F4-CH32V006F8/variants/CH32VM00X/README.md) this guide of how I did that fresh installation:

### Fresh installation with support for CH32VM00X (V002/V006/etc) ###

1. Install [latest WCH Link Utility](https://www.wch.cn/downloads/WCH-LinkUtility_ZIP.html), update LinkE device when prompted. Test connection to Link Utility and CH32 chip using Query Chip Info. (This step installs drivers and support for CH32V006 and other newer chips).
2. Install latest Arduino IDE (v2.3.6).
3. Run Arduino IDE as administrator, Add [board json](https://github.com/openwch/board_manager_files/raw/main/package_ch32v_index.json) to preferences. Install WCH CH32 core v1.0.4. Exit IDE.
4. Download [zip of latest master branch](https://github.com/openwch/arduino_core_ch32/archive/refs/heads/main.zip). Unzip and copy/overwrite all contents to this location: 
   - `C:\Users\<username>\AppData\Local\Arduino15\packages\WCH\hardware\ch32v\1.0.4`
5. Optional: review [unmerged PRs](https://github.com/openwch/arduino_core_ch32/pulls) for possibly required fixes and manually copy changed files.
6. Optional: for uploading to new chip: [update OpenOCD](https://github.com/maxint-rd/arduino_core_ch32/edit/support-CH32V002F4-CH32V006F8/variants/CH32VM00X/README.md#updating-openocd) obtained from MounRiver Studio.
7. Restart IDE. If new boards are not visible: [clear IDE application cache](https://github.com/maxint-rd/arduino_core_ch32/edit/support-CH32V002F4-CH32V006F8/variants/CH32VM00X/README.md#clearing-arduino-2x-ide-application-cache).
8. Select CH32 board. Test uploading simple serial example. Test debugging.