openwch/arduino_core_ch32 #49: CH32X035 board: Upload error even when upload is successful

smaharj1, 2024-01-14:
I followed all the instructions provided and am able to successfully compile and upload my CH32X035 through Arduino IDE. However, I get the upload error even when the upload seem to be working. I tested with multiple blinking patterns and uploaded them. The LED blinked according to my code. However, Arduino IDE gives an upload error. The port is the correct one too since I can see my code changes on the board. 

This is the error message I am seeing: 
```
Info : WCH-LinkE  mode:RV version 2.11 
Info : wlink_init ok
Info : clock speed 6000 kHz
Info : [wch_riscv.cpu.0] datacount=2 progbufsize=8
Info : [wch_riscv.cpu.0] Examined RISC-V core; found 1 harts
Info : [wch_riscv.cpu.0]  XLEN=32, misa=0x40901105
[wch_riscv.cpu.0] Target successfully examined.
Info : starting gdb server for wch_riscv.cpu.0 on 3333
Info : Listening on port 3333 for gdb connections
** Programming Started **
Info : device id = {someID}
Info : flash size = 62kbytes
** Programming Finished **
** Verify Started **
** Verified OK **

Failed uploading: uploading error: exit status 1
```

This is preventing me to use Serial Monitor because even if it uploads, I am getting the upload error.

0x0fe, 2024-01-26:
Hello,
The issue has been transmitted internally for review, meanwhile please confirm the following points : 
-is the code successfully uploaded to the CH32X035? you can verify this by adding a printf statement for example
-what is the version of your Arduino IDE

By the log it seems that the flashing was successful. Also i dont quite understand your last statement, the serial monitor is totally independant of the uploading, this is two different ports (uploading via SWD, serial monitor via CDC) so the LINKE can monitor the UART at any time, even during the upload. So normally there should be no problem to open the serial console even with this last error.

smaharj1, 2024-01-26:
- The code is being uploaded. I tested it with multiple `digitalWrite` to turn on the blue & green LEDs. 
- Arduino IDE version I tested with are 2.2.1 & 2.2.2-nightly-*. All of them give the same issue. 
- OS: mac OS. 

From what you are saying, I do see that the Serial Monitor is being connected. I tried the following piece of code to check if Serial Monitor is taking the input
```
String cmd;

void setup() {
  Serial.begin(115200);
  pinMode( LED_GREEN, OUTPUT );
  pinMode( LED_BLUE, OUTPUT );
}

void loop() {

  if (Serial.available()) {
    cmd = Serial.readString();
    digitalWrite(LED_BLUE, HIGH);

    if (cmd == "on") {
      digitalWrite(LED_GREEN, HIGH);
    } else if (cmd == "off") {
      digitalWrite(LED_GREEN, LOW);
    }
  } else {
    digitalWrite(LED_GREEN, HIGH);
  }
}
```
I get the Blue LED turned on, but the commands `on/off` that I sent are not doing anything.

maxint-rd, 2024-01-26:
Unfortunately the available() method is not implemented in this core. If you look at [HardwareSerial](https://github.com/openwch/arduino_core_ch32/blob/main/cores/arduino/HardwareSerial.cpp) you can see that available always returns the value -1. The peek() method also doesn't work. In my own version of this core I changed HardwareSerial and reintroduced the RX buffer to properly implement read/available/peek methods... 
My version isn't really tested other than on my own CH32V003, nor is the code up to any decent standards. It's too messy to share... Would be nice if WCH did pick this up properly though. (I already mentioned this in [an earlier post](https://github.com/openwch/arduino_core_ch32/issues/25#issuecomment-1756342813)...)

0x0fe, 2024-01-27:
@smaharj1 
So, the uploading works, you can ignore the error message. 

> to check if Serial Monitor is taking the input

Si it seems you are trying to read input from uart port? this hasnt been tested on our side and is indeed not implemented in the core. If you need thais function then you may want to grab the hardwareserial class from @maxint-rd  and refractor it into the official core.
This is however unrelated to this ticket.

DeqingSun, 2024-02-23:
Similiar issue on Mac OS 11.7.3 and Arduino 1.8.13.
So I did some test in terminal. It seems if OpenOCD get command "wlink_reset_resume;", the return will become 1 and Arduino will consider that as upload failure.

DeqingSun, 2024-02-29:
I tried a new version recommended from WCH support.

Using the openOCD executable inside 
http://file.mounriver.com/tools/MRS_Toolchain_MAC_V191.zip
can fix the upload error issue.

TianpeiLee, 2024-05-30:
Thank you for your attention. Upgrading to OpenOCD will solve the problem in the future

closed 2024-05-30 by TianpeiLee

DeqingSun, 2025-01-04:
http://www.mounriver.com/download
seems released a new version

http://file.mounriver.com/tools/MRS_Toolchain_MAC_V192.zip 

seems accessible.

dchicchon, 2025-04-25:
@DeqingSun the links you provided are not prompting download nor sending me to a download page for the tool. Any chance you could reupload the links?

DeqingSun, 2025-04-25:
@dchicchon click the first link to find the newest link to download.