openwch/arduino_core_ch32 #51: I2C Slave Support

Zeron, 2024-01-17:
Is I2C Slave supported in this core?

maxint-rd, 2024-02-11:
The code for slave support is in Wire.cpp. You can check out the [source code of the Wire library](https://github.com/openwch/arduino_core_ch32/tree/main/libraries/Wire/src). Have you tried it already? What were your findings?

Zeron, 2024-02-17:
I have been trying it and it does not respond to the master.

The below simple code works on other Arduino cores, but no on my CH32V003s

`#include <Arduino.h>
#include <Wire.h>

void RequestEvent() {

    static uint8_t Counter = 0;

    Serial.print(F("RequestEvent "));

    Wire.write(Counter);

    Counter++;

}

void ReceiveEvent(int NumBytes) {

    Serial.print(F("ReceiveEvent - NumBytes:"));
    Serial.println(NumBytes);

    for ( uint8_t i = 0; i < NumBytes; i++ ) {

        char c = Wire.read();
        Serial.print(c);

    }

    Serial.println();
    

}

void setup() {

    Serial.begin(115200);

    Serial.println(F("Stating at Address 0x22"));

    Wire.begin(0x22);
    
    Wire.onReceive(ReceiveEvent);

    Wire.onRequest(RequestEvent);

}

void loop() {
}`

maxint-rd, 2024-02-17:
Okay. If I get some spare time in the next few weeks, I'll see what I can reproduce with my CH32V003 and post my findings here.

kklam4697, 2024-02-19:
It seems the code had been commented
[image]

maxint-rd, 2024-02-19:
Oh great. looks like this is another area where the Arduino API isn't completely implemented. I already encountered something similar where I had to make some changes to fix Serial.available() and Serial.peek() and to get I2C scanning working on the CH32V003. Would be nice if an WCH employee could be dedicated to make this core fully compatible.

42retfa, 2024-03-16:
Hi,

Any news on I2C slave implementation for CH32V003 ? 

Thanks

maxint-rd, 2024-03-17:
@42retfa - not much. It doesn't work out of the box. WCH doesn't seem to give it much priority,

This weekend I did some experimenting and debugging on the CH32V003, but so far without usable results. When a slave address is specified the MCU can be scanned with an I2C scanner sketch. Using some core functions I managed to receive some data under iffy conditions. For now I haven't found the magic formula to sending data back in response to a read. Enough for this weekend... Perhaps more luck next time.

gitchrisha, 2024-03-19:
I agree with @maxint-rd, it would be great if someone from WCH was dedicated to this topic.
@TianpeiLee : do you have a roadmap on implementing I2C slave mode ? THANKS.

vanminh0910, 2024-03-19:
Slave I2C is my biggest reason to use these MCU as they are big saver on small cheap modules. Great if it can be supported.

42retfa, 2024-03-19:
> Slave I2C is my biggest reason to use these MCU as they are big saver on small cheap modules. Great if it can be supported.

I agree, the CH32 platform has the potential to become very popular within the Arduino community due to its capacity and low cost, but this requires that the most important functionalities to be supported : I2C slave mode is definitely one of them.

maxint-rd, 2024-03-19:
Eventually I did get something working for the v003, using functions and documentation found in the core system file ch32v00x_i2c.h. To make it all work somewhat more the Arduino way requires more effort, perhaps a few more weekends.
My own personal use case is an LCD module taken from a bank-card reader that needs plenty pins. I'd like to use the CH32V003 to provide the I2C interface and manage the display ram. To be continued...

maxint-rd, 2024-03-24:
FYI: It seems I have Wire slave functionality working; even managed to enable I2C interrupts. At the moment my code needs to be disabled when using master mode, so it's not quite production-ready yet. When I have something sharable I will publish it in my fork of this core. It also has some changes to support I2C-scanning.

BTW. Having debugging working was quite essential to get this far, so hats off for @DeqingSun who contributed PR #68 to make this happen. 

Latest hold-up was that somehow my I2C configuration application doesn't start when using -Os with LTO (Link Time Optimization), which is weird. It did work in Arduino IDE 1.8.19 with an older core, but in IDE v2.3.2 with core 1.4 it only works without LTO. Still need to test more variations and another sketch to see if my changes caused this. Does anyone else have issues with LTO?

maxint-rd, 2024-03-27:
Those who dare can try the Wire library in [this branch](https://github.com/maxint-rd/arduino_core_ch32/tree/I2C-slave-and-scanning) in my fork of the 1.0.4 core. 
I added basic examples for an I2C scanner and for a simple blinking slave.
*Work in progress* - **_USE AT YOUR OWN RISK_** - *No guarantees or support!* -

gitchrisha, 2024-03-28:
@maxint-rd : I just tested your Wire library in slave mode (send and receive) and everything works correctly. Thank you for your work which will allow me to move forward on my project. I hope it will be integrated into the official library.

maxint-rd, 2024-03-28:
Hello @gitchrisha, thank you for taking the effort to test and give feedback.
Good to read it works for you as well. I'm in no means an I2C expert so my code probably lacks some finesse and probably doesn't follow standards, especially in the area of error handling. I'm glad to contribute something working, that's better than nothing! 

After reading more about the CH32V003, I think it offers great hardware at such a low price and it would love to see this Arduino core get a bit more support, so we can use all its features...

maxint-rd, 2024-04-08:
Hello again,
For my CH32V003 I2C slave device I wanted to store some configurations, such as the I2C address. To support that I made this [EEPROM library](https://github.com/maxint-rd/arduino_core_ch32/tree/EEPROM-library/libraries/EEPROM), This release allows storing up to 26 bytes in the non-volatile user select word storage area. I don't have any of the other CH32 family members, so for now only the CH32V003 is supported. The library includes some documentation and Arduino examples to demonstrate its features.
Have fun!

Zeron, 2024-04-09:
Any know how to use https://github.com/maxint-rd/arduino_core_ch32/tree/I2C-slave-and-scanning via PlatformIO?

Zeron, 2024-04-09:
> Any know how to use https://github.com/maxint-rd/arduino_core_ch32/tree/I2C-slave-and-scanning via PlatformIO?

platform = https://github.com/Community-PIO-CH32V/platform-ch32v.git
platform_packages = 
    framework-arduino-openwch-ch32 = https://github.com/maxint-rd/arduino_core_ch32.git#I2C-slave-and-scanning

maxint-rd, 2024-04-10:
Did you get it working? I don't use platformio, but can't you just copy the entire Wire folder from my branch? https://github.com/maxint-rd/arduino_core_ch32/tree/I2C-slave-and-scanning/libraries/Wire

gitchrisha, 2024-04-16:
+1 with @maxint-rd. Just replace the Wire folder (/Users/xxxxxxx/.platformio/packages/framework-arduino-openwch-ch32/libraries in my OSX environment) with the maxint-rd Wire folder.

maxint-rd, 2024-06-19:
For readers: PR #108 containing this change was merged into the main branch. The branch mentioned above was consequently deleted.

inkeliz, 2025-01-19:
I didn't test it, yet. But, since this issue is "Open", that problem still exists or was fixed by https://github.com/openwch/arduino_core_ch32/pull/108?

maxint-rd, 2025-01-19:
The implementation in PR #108 was tested. a compilation issue was fixed by PR #126. Both PR's were merged into the master branch. As far as I know there were no further issues so I believe that this issue #51 can be closed. Unfortunately I don't have permissions to do it...

medtty, 2026-11-07:
I2C slave mode is in 1.0.5, see File → Examples → Wire → i2cMinimalSlave

closed 2026-11-07 by medtty