openwch/arduino_core_ch32 #56: Using other HW UARTs (3-6)

gpufler, 2024-01-29:
Hi,

first of all, thank you for the effort to implement this. 

I'm wondering how to use other UARTs/USARTs (I have boards that I need to use UART3-6) on PlatformIO, and I did not find a way to do it.

THX
Goran

P.S. 
I'm trying to implement some ethernet functionalities, and for now they work as normal classes, but I will try to catch time to implement it in Arduino  style (Ethernet, Client, ..) and if they work I send you a note to try to embed it into this. TXH

gpufler, 2024-02-12:
OK,
I solve it, I create UARTx.cpp (UART1.cpp to UART7.cpp) for all UARTs (I'm using CH32V307 with 8 UARTS), and I'm testing it now on my boards, so I can provide it if someone needs it.
I also adopt pins_arduino.h to define all pins of CH32V307 MCU and define all pins for UART's.
Now, I'm also adopting wchnet, but I have problems with interrupts, but I will open new thread for this.
Goran

yangliangquan, 2025-05-14:
In the project, only the serial port of the default Instance will be instantiated. For the remaining ones, you need to instantiate them yourself before use.

closed 2025-05-14 by yangliangquan