openwch/arduino_core_ch32 #57: Latest core doesn't build without warnings

maxgerhardt, 2024-01-31:
See below when warnings are actually enabled, for CH32X035G8U as selected board.
```
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\variants\CH32X035\CH32X035G8U\PeripheralPins.c:43:24: warning: excess elements in struct initializer
   {NC,        NP,   0, 0, 0}
                        ^
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\variants\CH32X035\CH32X035G8U\PeripheralPins.c:43:24: note: (near initialization for 'PinMap_ADC[12]')
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\variants\CH32X035\CH32X035G8U\PeripheralPins.c:43:27: warning: excess elements in struct initializer
   {NC,        NP,   0, 0, 0}
                           ^
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\variants\CH32X035\CH32X035G8U\PeripheralPins.c:43:27: note: (near initialization for 'PinMap_ADC[12]')
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\variants\CH32X035\CH32X035G8U\PeripheralPins.c:108:21: warning: excess elements in struct initializer
   {NC,   NP,     0, 0}
                     ^
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\variants\CH32X035\CH32X035G8U\PeripheralPins.c:108:21: note: (near initialization for 'PinMap_UART_TX[4]')
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\variants\CH32X035\CH32X035G8U\PeripheralPins.c:118:22: warning: excess elements in struct initializer
   {NC,    NP,     0, 0}
                      ^
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\variants\CH32X035\CH32X035G8U\PeripheralPins.c:118:22: note: (near initialization for 'PinMap_UART_RX[4]')
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\variants\CH32X035\CH32X035G8U\PeripheralPins.c:128:22: warning: excess elements in struct initializer
   {NC,    NP,     0, 0}
                      ^
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\variants\CH32X035\CH32X035G8U\PeripheralPins.c:128:22: note: (near initialization for 'PinMap_UART_RTS[4]')
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\variants\CH32X035\CH32X035G8U\PeripheralPins.c:138:22: warning: excess elements in struct initializer
   {NC,    NP,     0, 0}
                      ^
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\variants\CH32X035\CH32X035G8U\PeripheralPins.c:138:22: note: (near initialization for 'PinMap_UART_CTS[4]')
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\variants\CH32X035\CH32X035G8U\PeripheralPins.c:147:19: warning: excess elements in struct initializer
   {NC,   NP,   0, 0}
                   ^
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\variants\CH32X035\CH32X035G8U\PeripheralPins.c:147:19: note: (near initialization for 'PinMap_SPI_MOSI[1]')
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\variants\CH32X035\CH32X035G8U\PeripheralPins.c:154:19: warning: excess elements in struct initializer
   {NC,   NP,   0, 0}
                   ^
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\variants\CH32X035\CH32X035G8U\PeripheralPins.c:154:19: note: (near initialization for 'PinMap_SPI_MISO[1]')
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\variants\CH32X035\CH32X035G8U\PeripheralPins.c:161:19: warning: excess elements in struct initializer
   {NC,   NP,   0, 0}
                   ^
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\variants\CH32X035\CH32X035G8U\PeripheralPins.c:161:19: note: (near initialization for 'PinMap_SPI_SCLK[1]')
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\variants\CH32X035\CH32X035G8U\PeripheralPins.c:168:20: warning: excess elements in struct initializer
   {NC,    NP,   0, 0}
                    ^
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\variants\CH32X035\CH32X035G8U\PeripheralPins.c:168:20: note: (near initialization for 'PinMap_SPI_SSEL[1]')
Compiling .pio\build\genericCH32X035G8U6\FrameworkArduino\ch32\hw_config.c.o
Compiling .pio\build\genericCH32X035G8U6\FrameworkArduino\ch32\interrupt.cpp.o
Compiling .pio\build\genericCH32X035G8U6\FrameworkArduino\ch32\lib\ch32yyxx_adc.c.o
Compiling .pio\build\genericCH32X035G8U6\FrameworkArduino\ch32\lib\ch32yyxx_bkp.c.o
In file included from C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\cores\arduino\ch32\clock.c:13:
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\cores\arduino\ch32\backup.h: In function 'resetBackupDomain':
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\cores\arduino\ch32\backup.h:44:3: warning: implicit declaration of function 'RCC_BackupResetCmd'; did you mean 'RCC_AHBPeriphResetCmd'? [-Wimplicit-function-declaration]
   RCC_BackupResetCmd(ENABLE);
   ^~~~~~~~~~~~~~~~~~
   RCC_AHBPeriphResetCmd
Compiling .pio\build\genericCH32X035G8U6\FrameworkArduino\ch32\lib\ch32yyxx_can.c.o
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\cores\arduino\ch32\analog.cpp: In function 'uint16_t adc_read_value(PinName, uint32_t)':
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\cores\arduino\ch32\analog.cpp:548:12: warning: unused variable 'bank' [-Wunused-variable]
   uint32_t bank = 0;
            ^~~~
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\cores\arduino\ch32\analog.cpp: At global scope:
C:\Users\Max\.platformio\packages\framework-arduino-openwch-ch32\cores\arduino\ch32\analog.cpp:27:12: warning: 'calibration_value' defined but not used [-Wunused-variable]
 static int calibration_value = 0;
            ^~~~~~~~~~~~~~~~~
```

TianpeiLee, 2024-02-29:
thankyou, i will check it

maxint-rd, 2024-03-30:
Similar warning for CH32V003 after enabling ADC:
```
"C:\\Users\\..."..."...\\1.0.4\\variants\\CH32V00x\\CH32V003F4\\PeripheralPins.c"
C:\Users\...\1.0.4\variants\CH32V00x\CH32V003F4\PeripheralPins.c:39:24: warning: excess elements in struct initializer
   {NC,        NP,   0, 0, 0}
                        ^
C:\Users\...\1.0.4\variants\CH32V00x\CH32V003F4\PeripheralPins.c:39:24: note: (near initialization for 'PinMap_ADC[8]')
C:\Users\...\1.0.4\variants\CH32V00x\CH32V003F4\PeripheralPins.c:39:27: warning: excess elements in struct initializer
   {NC,        NP,   0, 0, 0}
                           ^
C:\Users\...\1.0.4\variants\CH32V00x\CH32V003F4\PeripheralPins.c:39:27: note: (near initialization for 'PinMap_ADC[8]')
```