openwch/arduino_core_ch32 #58: Error: error writing to flash at address 0x00000000 at offset 0x00000000

panciunio, 2024-02-02:
Hi,

Great job! I'm very happy to use Arduino with CH32VXXX but unfortunately I cannot upload proper compiled code to micro-controller:
```
Sketch uses 6028 bytes (9%) of program storage space. Maximum is 65536 bytes.
Global variables use 468 bytes (2%) of dynamic memory, leaving 20012 bytes for local variables. Maximum is 20480 bytes.
Open On-Chip Debugger 0.11.0+dev-02415-gfad123a16-dirty (2023-10-11-14:01)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'sdi'
Warn : Transport "sdi" was already selected
Ready for Remote Connections
Info : WCH-LinkE  mode:RV version 2.9 
Info : wlink_init ok
Info : clock speed 6000 kHz
Info : [wch_riscv.cpu.0] datacount=2 progbufsize=8
Info : [wch_riscv.cpu.0] Examined RISC-V core; found 1 harts
Info : [wch_riscv.cpu.0]  XLEN=32, misa=0x40901105
[wch_riscv.cpu.0] Target successfully examined.
Info : starting gdb server for wch_riscv.cpu.0 on 3333
Info : Listening on port 3333 for gdb connections
** Programming Started **
Info : device id = 0xd56babcd
Info : flash size = 64kbytes
Error: error writing to flash at address 0x00000000 at offset 0x00000000
embedded:startup.tcl:1162: Error: ** Programming Failed **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 1227
at file "embedded:startup.tcl", line 1162
Failed uploading: uploading error: exit status 1
 
```

Where I should start investigation? Especially, there is no an issue when I trying to flash e.g. directly from VisualStudioCode.

I'm working under LinuxMint 23.4
ArduinoIDE: 2.2.1
CH32 MCU EVT Boards: 1.0.4

Tested on two pieces of nanoCH32V203 (CH32V203C8T6)

maxgerhardt, 2024-02-04:
>Info : WCH-LinkE  mode:RV version 2.9 
>Error: error writing to flash at address 0x00000000 at offset 0x00000000

I have had this **exact** error and this is the updated OpenOCD version is incapable of speaking with a WCH-LinkE of an older firmware version. You're running v2.9 but it needs v2.11. See https://github.com/Community-PIO-CH32V/platform-ch32v?tab=readme-ov-file#important-notices on how to update.

petrkr, 2024-04-17:
I can confirm after update FW to 2.12 it works fine.

Update is possible only at windows, so I had to run Virtual Machine.

maxgerhardt, 2024-04-17:
Closing since this is a confirmed fix and we have instructions for that in the README now.

TianpeiLee, 2024-05-29:
> Hi,
> 
> Great job! I'm very happy to use Arduino with CH32VXXX but unfortunately I cannot upload proper compiled code to micro-controller:
> 
> ```
> Sketch uses 6028 bytes (9%) of program storage space. Maximum is 65536 bytes.
> Global variables use 468 bytes (2%) of dynamic memory, leaving 20012 bytes for local variables. Maximum is 20480 bytes.
> Open On-Chip Debugger 0.11.0+dev-02415-gfad123a16-dirty (2023-10-11-14:01)
> Licensed under GNU GPL v2
> For bug reports, read
> 	http://openocd.org/doc/doxygen/bugs.html
> Info : only one transport option; autoselect 'sdi'
> Warn : Transport "sdi" was already selected
> Ready for Remote Connections
> Info : WCH-LinkE  mode:RV version 2.9 
> Info : wlink_init ok
> Info : clock speed 6000 kHz
> Info : [wch_riscv.cpu.0] datacount=2 progbufsize=8
> Info : [wch_riscv.cpu.0] Examined RISC-V core; found 1 harts
> Info : [wch_riscv.cpu.0]  XLEN=32, misa=0x40901105
> [wch_riscv.cpu.0] Target successfully examined.
> Info : starting gdb server for wch_riscv.cpu.0 on 3333
> Info : Listening on port 3333 for gdb connections
> ** Programming Started **
> Info : device id = 0xd56babcd
> Info : flash size = 64kbytes
> Error: error writing to flash at address 0x00000000 at offset 0x00000000
> embedded:startup.tcl:1162: Error: ** Programming Failed **
> in procedure 'program' 
> in procedure 'program_error' called at file "embedded:startup.tcl", line 1227
> at file "embedded:startup.tcl", line 1162
> Failed uploading: uploading error: exit status 1
>  
> ```
> 
> Where I should start investigation? Especially, there is no an issue when I trying to flash e.g. directly from VisualStudioCode.
> 
> I'm working under LinuxMint 23.4 ArduinoIDE: 2.2.1 CH32 MCU EVT Boards: 1.0.4 Programmer: WCH-inkE-R0 v1.3
> 
> Tested on two pieces of nanoCH32V203 (CH32V203C8T6)
> 
> EDIT1, I checked the same program and programmer with nanoCH32V003 and everything works fine:
> 
> ```
> riscv-none-elf-g++: warning: '-x c++' after last input file has no effect
> Found WCH Link
> WCH Programmer is LinkE version 2.8
> Chip Type: 003
> Part Type (A): 0x0010 (This is the capacity code, in KB)
> Part UUID    : 7f-e2-ab-cd-e8-13-bc-64
> PFlags       : ff-ff-ff-ff
> Part Type (B): 00-31-05-00
> Interface Setup
> Image written.
> ```
> 
> EDIT2, The same probe as above I tried with CH32V103 (C8T6) and result is the same as for x203:
> 
> ```
> Sketch uses 5316 bytes (8%) of program storage space. Maximum is 65536 bytes.
> Global variables use 476 bytes (2%) of dynamic memory, leaving 20004 bytes for local variables. Maximum is 20480 bytes.
> Open On-Chip Debugger 0.11.0+dev-02415-gfad123a16-dirty (2023-10-11-14:01)
> Licensed under GNU GPL v2
> For bug reports, read
> 	http://openocd.org/doc/doxygen/bugs.html
> Info : only one transport option; autoselect 'sdi'
> Warn : Transport "sdi" was already selected
> Ready for Remote Connections
> Info : WCH-LinkE  mode:RV version 2.9 
> Info : wlink_init ok
> Info : clock speed 6000 kHz
> Info : [wch_riscv.cpu.0] datacount=2 progbufsize=8
> Info : [wch_riscv.cpu.0] Examined RISC-V core; found 1 harts
> Info : [wch_riscv.cpu.0]  XLEN=32, misa=0x40101105
> [wch_riscv.cpu.0] Target successfully examined.
> Info : starting gdb server for wch_riscv.cpu.0 on 3333
> Info : Listening on port 3333 for gdb connections
> ** Programming Started **
> Info : device id = 0x9ba6abcd
> Info : flash size = 64kbytes
> Error: error writing to flash at address 0x00000000 at offset 0x00000000
> embedded:startup.tcl:1162: Error: ** Programming Failed **
> in procedure 'program' 
> in procedure 'program_error' called at file "embedded:startup.tcl", line 1227
> at file "embedded:startup.tcl", line 1162
> Failed uploading: uploading error: exit status 1
> ```
@panciunio May I ask if you are under macOS? It seems that there is an issue with the return value judgment of a function in openocd under macOS. However, wch has been updated, and the repository will also be updated in the future. 

I'm sorry, I am the only one who maintains the repository in my spare time, and the updates may not be particularly timely. However, I will try my best to solve as many problems as possible when adding new features or boards

closed 2025-05-14 by yangliangquan