openwch/arduino_core_ch32 #59: Ch32X033 EVT Board can't work

ziv2013, 2024-02-04:
I have got a Ch32X033 EVT board from WCH. And it works well with MountRiver.

But I don't know why it can't work with arduino_core_ch32.

maxgerhardt, 2024-02-04:
Any additional info? Does compiling fail? Does uploading fail? Does the uploaded sketch not work? What code are you using, what Arduino IDE "Tools" settings are you using?

ziv2013, 2024-02-05:
> Any additional info? Does compiling fail? Does uploading fail? Does the uploaded sketch not work? What code are you using, what Arduino IDE "Tools" settings are you using?

Thanks a lot. Your comments give me more confidence. I make an experiment with a GPIO.

Ch32X033 is workable to driven PA_0.

closed 2024-02-05 by ziv2013

maxint-rd, 2024-10-11:
FYI. I'm currently working on a board variant for the CH32X033F8P6 that should have many more Arduino pins working properly.