openwch/arduino_core_ch32 #60: WCHNET library integration

gpufler, 2024-02-12:
Hi,
I'm trying to implement wchnet library in Arduino framework on platformIO, but I run into a problems.
So, I succeed to add wchnet as library in platformIO, and it compile without problem, but it's not working.
As far as I can trace, problem is in interrupts. MCU freeze on TIM2 initialization and if I remove TIM2, ETH_IRQHandler does not start and ETH_LibInit does not start.
it probably collide with some previous initialization of timers, but I can't find which one.

After I make this work I plan to make complete implementation of Ethernet.cpp so that this will be compatible with other Arduino libraries. 

Can you help me with this?

THX
Goran

maxgerhardt, 2024-02-15:
The WCHnet library works in the PlatformIO integration with the CH32V307 example. I have personally tested https://github.com/Community-PIO-CH32V/platform-ch32v/tree/develop/examples/webserver-ch32v307-none-os and documented the results.

I also had the same idea of pulling in the wchnet library, or adding in the lwIP stack in source code form + adding the integration interface between the ethernet hardware and lwIP.

maxgerhardt, 2024-02-16:
I see. I'll look into this for ch32v307, The `millis()` code uses the SysTick timer of the CPU, PWM uses [TIMx timers](https://github.com/openwch/arduino_core_ch32/blob/main/cores/arduino/ch32/analog.cpp#L682-L705), but usually nothing should `analogWrite()` by default.

gpufler, 2024-02-16:
I succeed to track it to` NVIC_EnableIRQ(TIM2_IRQn);` line, where he freezes. But I noticed something odd, I see that in this repository there is a wiring_analog and other analog files, but in my installation, there is none of this.  I use some old version; I will try to reinstall.