openwch/arduino_core_ch32 #62: 'yield' was not declared in this scope

kklam4697, 2024-02-14:
HI
I was tring to use this framework in my CH32V003 borad. 
How every When I try to use U8G2 library, I found error when I try to verify. 

**U8g2lib.cpp:47:3: error: 'yield' was not declared in this scope**
[image]

I found yeild function was commented on Arduino.h file
[image]

I try to uncomment it, And give it a empty implementation. Then will complie pass.
I also try comment the yeild function call on U8G2 source code, Then will also pass. 

But both this two method can be use in VScode+PlatformIO, That would be so diffical to adopt on Arduino IDE. 

Can you please add a yeild function implementation in the Core? Thanks a lot

maxint-rd, 2024-02-14:
Upvote for yield(0. See also issue #21

kklam4697, 2024-02-15:
> Upvote for yield(0. See also issue #21

Thanks for the reffenence. Maybe they will response after Chinese new Year.

TianpeiLee, 2024-03-04:
i have add yield(default weak) function into core library

closed 2024-03-04 by TianpeiLee