openwch/arduino_core_ch32 #63: Flash example

silvalange, 2024-02-16:
Hello!
Any predictions for an example implementation for using flash?  something similar to what we have in ESP which is an emulated EEPROM.

rodrigo-s-lange, 2024-02-20:
It is not clear from the flash example how I can read a byte after the system restarts. Has anyone managed to do something similar? I just needed to save a byte and read it after restarting ch32

rodrigo-s-lange, 2024-02-25:
resolvido!

maxint-rd, 2024-04-02:
Hello @rodrigo-s-lange - You didn't write how you resolved this or which example you were talking about...

Nevertheless;  today I experimented a bit with my CH32V003 and code from the [optiondata example](https://github.com/cnlohr/ch32v003fun/tree/master/examples/optiondata) by @cnlohr.
After copying the required FLASH Keys and Flash Control Register bits definitions from [ch32v00x_flash.c](https://github.com/openwch/arduino_core_ch32/blob/main/system/CH32V00x/SRC/Peripheral/src/ch32v00x_flash.c) the example worked and successfully showed how a value was preserved after reset.

Based on that example I think it's possible to make a (limited) EEPROM library, comparable to the eeprom emulation done for the ESP8266/ESP32. From what I read in the datasheet, the CH32V003 has a 64-byte area for option bytes, some of which already used, but some potentially available for application purposes. For my own project this number would be sufficient to store some settings...

maxint-rd, 2024-04-05:
Done. :-)
Today I published a first release of an [EEPROM library](https://github.com/maxint-rd/arduino_core_ch32/tree/EEPROM-library), This release supports storing up to 26 bytes in the 
non-volatile the user select word storage area. I don't have any of the other CH32 family members, so for now only the CH32V003 is supported. 
This release is based on the Option Data example by @CNLOHR and other Arduino EEPROM implementations. It includes some documentation and Arduino examples to demonstrate its features.

rodrigo-s-lange, 2024-04-05:
> Hello @rodrigo-s-lange - You didn't write how you resolved this or which example you were talking about...
>
> Nevertheless; today I experimented a bit with my CH32V003 and code from the [optiondata [...]

Hello
I solved it using a method that was not in HAL's examples.
I had to send an email to the CH people. They responded quickly to the email.
I used the reserved part of the memory.
And to read something like reading the address directly:
*(u32*)0x1FFFF80C

rodrigo-s-lange, 2024-04-05:
> Done. :-) Today I published a first release of an [EEPROM library](https://github.com/maxint-rd/arduino_core_ch32/tree/EEPROM-library/libraries/EEPROM), This release supports storing up to 26 bytes [...]

Great job!
I'm testing your examples.
It was very complete.
The information for CH32 is still inconsistent. Some libs don't work very well and we had to make some fixes.
I still haven't been able to use libfun without the openWCH platform.
And some bugs occur, even in some cases I can't reference libfun

maxint-rd, 2024-04-05:
Hi @rodrigo-s-lange, thank you for your feedback! 
> I solved it using a method that was not in HAL's examples. I had to send an email to the CH people. They responded quickly to the email.

Interesting to read you got a swift reply while on the other hand WCH seems to leave this Arduino core and the contributions kind of dangling. I guess it's about priorities...

> I used the reserved part of the memory. And to read something like reading the address directly: *(u32*)0x1FFFF80C

For other readers: That particular address is mentioned at the top of page 179 of the CH32V003 Reference Manual. At the bottom of that page it's stated to be the reserved byte for WRP2, one of the 8 user-option bytes that are mentioned on page 178:
> _"The user-option bytes is solidified in FLASH and will be reloaded into the corresponding register after system reset, and can be erased and programmed by the user at will. The user option bytes information block has a total of 8 bytes (4 bytes for write protection, 1 byte for read protection, 1 byte for configuration options, and 2 bytes for storing user data), and each bit has its inverse code bit for checksum during loading."_

Because these are loaded after system reset into their register, one can read that byte from a memory address. To write it one needs to use the WCH LinkUtility or the some other code that erases the 64 byte flash page and rewrite the other option bytes.

Although the WRP2 and WRP3 bytes seem unused for the V003, they are listed as reserved and so I decided to skip them in my EEPROM library. Next to the data0 and data1 bytes that are labeled for storing user data, my EEPROM implementation uses the rest of the 64 byte page to get 24 more words that can contain data bytes and their inverse. 26 bytes is enough for me for now...

> The information for CH32 is still inconsistent. Some libs don't work very well and we had to make some fixes. 

Good to know. As a maker I'm using several different processors in very low quantities. I enjoy using them in a common Arduino environment and find the CH32 family a welcome addition that offers great value for money. I wish this CH32 Arduino core gets a bit more attention. I also had to make several fixes to make it more useful for my CH32V003 project (e,g. fixing Serial.available(), completing I2C slave and enabling scanning). I needed this EEPROM feature to allow my I2C slave device respond to a configuration tool and  store some settings.

> I still haven't been able to use libfun without the openWCH platform. And some bugs occur, even in some cases I can't reference libfun
As I prefer to stick to a single environment I haven't used ch32v003fun directly, but was happy to find out I could use the Option Data example within this Arduino core. I find Charles' work quite amazing, original, dedicated and very inspiring. As I enjoy using his work and that of many other people. I hope my contribution is likewise helpful to others...

medtty, 2026-11-07:
1.0.5 includes the EEPROM library with examples: File → Examples → EEPROM

closed 2026-11-07 by medtty