openwch/arduino_core_ch32 #64: CH32X035 I2C is disabled. Why?

kklam4697, 2024-02-19:
HI I found CH32X035 I2C is disabled. If uncomment it, the will pass complie(Not yet download to the bord).
May I know if any bug found for CH32X035 I2C? Thanks

[image]

[image]

maxint-rd, 2024-02-19:
Some time ago another user encountered this too. I had the same with SPI on the CH32V003. See [this issue](https://github.com/openwch/arduino_core_ch32/issues/29#issuecomment-1840733085).

maxgerhardt, 2024-02-21:
Are you sure that you have won the silicon lottery? See the X035 datasheet.

[image]

kklam4697, 2024-02-22:
> Are you sure that you have won the silicon lottery? See the X035 datasheet.
> 
> ![grafik](https://private-user-images.githubusercontent.com/26485477/306721531-e2a6e9ad-6896-4e12-8a04-34c80f4d2769.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg1NjUxNTAsIm5iZiI6MTcwODU2NDg1MCwicGF0aCI6Ii8yNjQ4NTQ3Ny8zMDY3MjE1MzEtZTJhNmU5YWQtNjg5Ni00ZTEyLThhMDQtMzRjODBmNGQyNzY5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjIyVDAxMjA1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA3MzNmNTExNzIwYjM4ZWFlMGNlNDBkNjM1OTc3MDNlZTUxMDczZWQ5ODUyZDVkZTg3ODQxZjg4Yzg5M2Y2ZWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.mDR-l2ratcRFpTigYdqF5zemyG6un3cM4HTgogFVoDg)

HI, I also found that issue. I just call WCH, They told me for newer patch of chip will not have that issue. We should not able to buy such old patch of chip in the market

TianpeiLee, 2024-02-29:
To verify this function, a corresponding version of the chip needs to be used here.

closed 2024-02-29 by TianpeiLee