openwch/arduino_core_ch32 #77: CH32X035 compilation broken in HardwareTimers.cpp

maxgerhardt, 2024-03-27:
The lines

https://github.com/openwch/arduino_core_ch32/blob/c40c5c8095060c523e635b99426314fab56fe12c/cores/arduino/HardwareTimer.cpp#L72-L74

fail to compile when selecting a CH32X035 board.

```
/home/runner/.platformio/packages/framework-arduino-openwch-ch32/cores/arduino/HardwareTimer.cpp:73:18: error: 'TIM2_IRQn' was not declared in this scope
   NVIC_EnableIRQ(TIM2_IRQn);
                  ^~~~~~~~~
/home/runner/.platformio/packages/framework-arduino-openwch-ch32/cores/arduino/HardwareTimer.cpp:73:18: note: suggested alternative: 'TIM3_IRQn'
   NVIC_EnableIRQ(TIM2_IRQn);
                  ^~~~~~~~~
                  TIM3_IRQn
```

This is because the CH32X035 **does** define `TIM2_BASE` macro but doesn't have a `TIM2_IRQn` interrupt. 

https://github.com/openwch/arduino_core_ch32/blob/c40c5c8095060c523e635b99426314fab56fe12c/system/CH32X035/SRC/Peripheral/inc/ch32x035.h#L575-L576

It has these though

https://github.com/openwch/arduino_core_ch32/blob/c40c5c8095060c523e635b99426314fab56fe12c/system/CH32X035/SRC/Peripheral/inc/ch32x035.h#L65-L81

maxgerhardt, 2024-03-27:
I have attempted a fix for that in https://github.com/Community-PIO-CH32V/arduino_core_ch32/commit/bbf32d397644cac7c465f0770fec8ed44cb14e0a, for reference.

medtty, 2026-10-30:
Builds on current main: X035 compiles with HardwareTimer on TIM2 and TIM3 (tested on a CH32X035). please reopen if you still see it

closed 2026-10-30 by medtty