openwch/arduino_core_ch32 #83: Delay_Us() bug

ziv2013, 2024-04-19:
I have tried Delay_Us() with Ch32V305. There is a bug which would cause millis() goes wrong.

E.g.,  I write a code as below

void setup() {
  //GPIO_InitTypeDef GPIO_InitStructure = {0};
  //Initialize serial and wait for port to open:
  Serial.begin(115200);
  Serial.println("start");
}

void loop() {
  Serial.println(millis());
  //delay(1000);    //This works well
  Delay_Us(1000000);//This can't work  
}

I would just got 

"start
0"

in serial.

tinatsina, 2024-04-22:
## Wrong Function Used 🛑
- please use the `delayMicroseconds()`
- you can find more information here [Arduino CC Ref](https://www.arduino.cc/reference/en/language/functions/time/delaymicroseconds/)

ziv2013, 2024-04-22:
OK. You are right. Delay_us() is not a standard function of Arduino.

> ## Wrong Function Used 🛑
> * please use the `delayMicroseconds()`
> * you can find more information here [Arduino CC Ref](https://www.arduino.cc/reference/en/language/functions/time/delaymicroseconds/)

closed 2024-04-22 by ziv2013