openwch/arduino_core_ch32 #84: Demo code for CH32V307 USB Keyboard

ziv2013, 2024-04-22:
Hello All，

I have written a demo code for CH32V307 USB Keyboard function and verify it with Ch32V305.

[CH32X30XKBMSHS.zip](https://github.com/openwch/arduino_core_ch32/files/15057751/CH32X30XKBMSHS.zip)

You can try it if you are interest in this function

Z.t

TianpeiLee, 2024-05-30:
I think the repository needs a unified USB abstraction layer instead of targeting specific models, otherwise adding chips in the future will be more troublesome. If you have already ported it, you can submit a PR

closed 2024-05-30 by TianpeiLee

ziv2013, 2024-05-30:
> I think the repository needs a unified USB abstraction layer instead of targeting specific models, otherwise adding chips in the future will be more troublesome. If you have already ported it, you can submit a PR

I just write code for this chip and it's too hard for me to make  a unified USB abstraction layer. It's far beyond my ability. But as USB is the advantage of WCH chips. I suggested to add the official support for USB.