openwch/arduino_core_ch32 #85: tools/atomic.h not found

Styne13, 2024-05-02:
Hi,

thank you for the great work on this project.

I try to use ch32v003 with the radiohead library to connect to ASK or Lora RF modules. If I try to compile the code, I get _tools/atomic.h_ not found. I had a look and it is normally located in the avr code for original arduino, so I would guess the core ch32 shall adapt it somehow to be compatible.

maxint-rd, 2024-05-04:
Have you tried [this release](https://github.com/epsilonrt/RadioHead)? It contains a folder named RHutil, which also contains [atomic.h](https://github.com/epsilonrt/RadioHead/blob/master/RHutil/atomic.h).

Styne13, 2024-05-07:
Thank you for your advice. 👍
I will try out this fork of the library.

Styne13, 2024-05-07:
Ok, looked into the code of the library and RHutil/atomic.h is only used in the case of __arm__
If I try to include it in else path, other things like ATOMIC_BLOCK are not declared.

TianpeiLee, 2024-05-30:
@Styne13 Can you provide the function declaration for the atomic operation you need so that I can know what functions I need to provide? The AMO extension of RISC-V provides the following instructions:
[amo.zip](https://github.com/openwch/arduino_core_ch32/files/15494693/amo.zip)

fearpro13, 2024-10-13:
@TianpeiLee 
Encountered same problem, missing atomic.h for RadioHead library

Function definitions that needs to be provided are here:
https://github.com/epsilonrt/RadioHead/blob/master/RHutil/atomic.h

FUFU9491, 2025-05-14:
@Styne13 
I saw this topic for a long time, I don't know if you have done it yet. I am using CH32v0003 and radiohead library and got the same error as you, if you have done it successfully, please guide me. Thank you very much.

closed 2026-10-30 by medtty