openwch/arduino_core_ch32 #87: Upload Failing Windows 10 (WCH-LinkE)

prairielandelec, 2024-05-10:
Just got a CH32C0003F4P6-Ro-1v1 and a WCH-LinkE-Ro-1v1 and am attempting to upload a test sketch.

First I installed the board packed per the instructions in this repo

I have connected SWDIO to PD1 on the dev board as well as 3v3 and GND and RX to PD5

When I look at Device manager there is a USB Serial Device and an unknown device WCH-Link

I select the USB Serial Device as the port in the Arduino IDE and the CH32V00x as the board and put in a simple blink test.

When I click "upload" I get the following:
```
Sketch uses 4796 bytes (29%) of program storage space. Maximum is 16384 bytes.
Global variables use 396 bytes (19%) of dynamic memory, leaving 1652 bytes for local variables. Maximum is 2048 bytes.
Open On-Chip Debugger 0.11.0+dev-02415-gfad123a16-dirty (2023-05-05-13:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'sdi'
Warn : Transport "sdi" was already selected
Ready for Remote Connections
Failed uploading: uploading error: exit status 0xc0000005
```

Is this just a faulty WCH-LinkE? Or am I missing some step of the initial configuration process? The bin builds just fine but after this error shows up nothing happens.

Thanks!

maxint-rd, 2024-05-11:
I think you're missing some driver. In my Windows 11 system I see both a USB Serial COM port and under Interfaces a WCH-LinkRV device. The latter I believe handles the SWIO pin for programming and real-time debugging, while the COM port allows showing output from Serial.println() and input via Serial.read().
(For flashing the chip the debugging port is used).

prairielandelec, 2024-05-11:
It would appear that is the case, I could not find any instructions in this repo and scant documentation in general. My understanding was that the WCH-LinkE utilized the OpenOCD drivers to some degree. I was looking through the Platoform IO Platform for this chip and saw the following section of their documentation:
https://pio-ch32v.readthedocs.io/en/latest/installation.html#install-drivers-rules

I tried this driver on a spare laptop I had and it seemed to get things working. Biggest difference was seeing the WCH-Link interface appear in the "interfaces" section of device manager. I was able to program the following to the chip:

```#define LED_PIN D5

void setup() {
  pinMode(LED_PIN, OUTPUT);
}

void loop() {
  digitalWrite(LED_PIN, HIGH);
  delay(100);
  digitalWrite(LED_PIN, LOW);
  delay(100);
}
```

The verification check seemed to pass just fine, however I do not see the output toggling on any of the pins. Is there any documentation on the pin naming? From what I can tell from other examples I have configured everything correctly.

Just trying to pinpoint if this is a problem with the flashing process, the code, or the chip on my dev board.

Thanks for the help!

maxint-rd, 2024-05-11:
There are some cheat sheets that show the pinouts of the various chips. For the F4P6 I made a PDF out of the image I found somewhere: [ch32v003f4p6_pinout.pdf](https://github.com/openwch/arduino_core_ch32/files/15283568/ch32v003f4p6_pinout.pdf)

Usually I use pin names such as PC1, PA1, etc.
Have fun!

TianpeiLee, 2024-05-30:
When using wch-linke, it is important to ensure that the driver for wch-linke is installed correctly and that its firmware version is the latest. This can be done through the "WCH LinkUtility" tool to install the driver and update the firmware.
https://www.wch.cn/downloads/WCH-LinkUtility_ZIP.html

In addition to using pinout diagrams similar to those provided by maxint-rd,  there is also an imperfect way  to query through the file "variantCH32V003F4.cpp" and the RM

closed 2024-05-30 by TianpeiLee

ITstreet1, 2025-08-12:
I have a similar problem, but with the CH32L103C8T6. The WCH-LinkE firmware is up to date. While I try to upload a simple Blink sketch, I get:

Sketch uses 7896 bytes (12%) of program storage space. Maximum is 65536 bytes.
Global variables use 476 bytes (2%) of dynamic memory, leaving 20004 bytes for local variables. Maximum is 20480 bytes.
Open On-Chip Debugger 0.11.0+dev-02415-gfad123a16-dirty (2023-05-05-13:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'sdi'
Warn : Transport "sdi" was already selected
Ready for Remote Connections
Info : WCH-LinkE  mode:RV version 2.12 
Error:  communication fail,please contact [support@mounriver.com]

Failed uploading: uploading error: exit status 1

What to check?

maxint-rd, 2025-08-12:
Can you try the WCH Link Utility? Using the utility you can connect and read chip information. If that succeeds, you should also be able to use features such as Power on erase and also program it using the binary that the Arduino IDE generated. 
If reading chip info fails, you need to make sure you selected the correct chip and check your connections.
If needed update that WCH Link Utility to the latest version also.

Unfortunately I don't own any 10x chips, so I can't help you in testing things.

maxint-rd, 2025-08-13:
Good to know that updating OpenOCD made some difference. 
Thanks for sharing your findings. That may come in handy for others with similar setup.

It's quite a list of messages you got. In my most recent tests with V002/V003/V006 and X033, I sometimes got one or two messages _after_ successful verification. Never such a long list of errors before that OK message. Unfortunately I don't have a CH32L103 (or V103) to check your findings, so I can't confirm. @KenRouKoro did some recent testing on his V103, but didn't mention such errors. They all seem to relate to registers, perhaps their values are different from expected. 

If I were you I would now test these things:
- check the wires to see if they cause noise. In my latest tests I used a flaky breadboard with rather long jumper wires. Sometimes uploading failed and the Link Utility didn't recognize the chip. Reinserting SWIO/SWD fixed it. 
- Use the Link Utility to upload the generated binary and see if the verify step reports any errors
- Try out the real-time debugging within the Arduino IDE to check if it reports any errors. (If no errors: happy debugging and developing!)

KenRouKoro, 2025-08-13:
I have tested 10cm and 20cm DuPont cables. The 20cm DuPont cable will completely fail to program (verification failed) for a period of time (it's very sudden that there will be a period of time), and then it will be permanently unable to program, at which point the WCHLinkE needs to be re-plugged. However, the 10cm one has none of these problems.

maxint-rd, 2025-08-13:
Okay. Then assuming the connection is solid, I guess you would need to try programming another bare chip that has no other passives connected, to see if the messages are specific to your board.

Perhaps a good next step is to see if the Link Utility reports any errors in the verification step, and then to see if debugging is working without errors. Good luck!