openwch/arduino_core_ch32 #88: Upload error - "Read-Protect Status Currently Enabled" on new/blank EVT

beehphy, 2024-05-11:
Testing out the simplest code just to explore programming the EVT in Arduino IDE 2.3.2. 
Got the WchLink programmer going with Ard.IDE by installing the WHQL driver found inside this ZIP: https://www.wch.cn/downloads/WCH-LinkUtility_ZIP.html

```
void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.print("Bark\n");
  delay(200);
}
```

Code compiles and when it goes to program there is an error:

```
Open On-Chip Debugger 0.11.0+dev-02415-gfad123a16-dirty (2023-05-05-13:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'sdi'
Warn : Transport "sdi" was already selected
Ready for Remote Connections
Info : WCH-LinkE  mode:RV version 2.12 
Info : wlink_init ok
Info : clock speed 6000 kHz
Info : [wch_riscv.cpu.0] datacount=2 progbufsize=8
Info : [wch_riscv.cpu.0] Examined RISC-V core; found 1 harts
Info : [wch_riscv.cpu.0]  XLEN=32, misa=0x40901105
[wch_riscv.cpu.0] Target successfully examined.
Info : starting gdb server for wch_riscv.cpu.0 on 3333
Info : Listening on port 3333 for gdb connections
** Programming Started **
Info : device id = 0x06beabcd
Info : flash size = 62kbytes
Error: Read-Protect Status Currently Enabled
Error: failed erasing sectors 0 to 5
embedded:startup.tcl:1162: Error: ** Programming Failed **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 1227
at file "embedded:startup.tcl", line 1162
Failed uploading: uploading error: exit status 1
```

Seems like the path to the chip is open but something else is busted. 
Do I need to install more of the programs found in the WchLink Zip?
Any help appreciated.

maxint-rd, 2024-05-11:
What info do you get from the Link utility? Is your programmer a WCH LinkE model? Is the programmer in  RISC-V mode?
Can you use all features of that utility? (Query Chip Info, Query RW Protect Status, Read Chip Flash, Reset)

michpro, 2024-05-11:
Try to unlock the chip using [WCH-LinkUtility](https://www.wch.cn/downloads/WCH-LinkUtility_ZIP.html).
[image]

beehphy, 2024-05-11:
Attempting to unlock the chip results in what looks like a success:
[image]

logs from Arduino IDE:
`Open On-Chip Debugger 0.11.0+dev-02415-gfad123a16-dirty (2023-05-05-13:43)
```
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'sdi'
Warn : Transport "sdi" was already selected
Ready for Remote Connections
Info : WCH-LinkE  mode:RV version 2.12 
Info : wlink_init ok
Info : clock speed 6000 kHz
Info : [wch_riscv.cpu.0] datacount=2 progbufsize=8
Info : [wch_riscv.cpu.0] Examined RISC-V core; found 1 harts
Info : [wch_riscv.cpu.0]  XLEN=32, misa=0x40901105
[wch_riscv.cpu.0] Target successfully examined.
Info : starting gdb server for wch_riscv.cpu.0 on 3333
Info : Listening on port 3333 for gdb connections
** Programming Started **
Info : device id = 0x06beabcd
Info : flash size = 62kbytes
** Programming Finished **
** Verify Started **
** Verified OK **`

```

maxint-rd, 2024-05-11:
Excellent. Congratulations! 
Trying the other link utility functions such as Query Chip Info, Read Chip Flash, Reset can also be useful. 

Have fun programming!