openwch/arduino_core_ch32 #89: CH32 v1.04 doesn't install on Windows 10

cdemos, 2024-05-11:
The 1.04 update won't install on Arduino IDE 2.3.2. Version 1.03 installs fine and seems to work. I get the following message when trying to install 1.04 on Windows 10:

```
Downloading packages
WCH:beforeinstall@1.0.0
WCH:ch32v@1.0.4Installing WCH:beforeinstall@1.0.0
Failed to install platform: 'WCH:ch32v:1.0.4'.
Error: 13 INTERNAL: Cannot install tool WCH:beforeinstall@1.0.0: extracting archive: Create link
C:\Users\me\AppData\Local\Arduino15\tmp\package-2892474692\wch_beforeinstall_linux-1.0.0\libhidapi-hidraw.so: 
symlink libhidapi-hidraw.so.0.0.0 C:\Users\me\AppData\Local\Arduino15\tmp\package-2892474692\wch_beforeinstall
_linux-1.0.0\libhidapi-hidraw.so: A required privilege is not held by the client.
```
I also get the same error on a different PC running Windows 11. Looking at other postings, it seems like an error in the install script.

maxint-rd, 2024-05-11:
Might be related to issue #44 - have you tried to run the installer as Administrator? That worked for me on Windows 11.

cdemos, 2024-05-13:
It looked a bit more complicated than I’d like, and since 1.03 seems fine, I’ll hold off for now and hopefully it will it be fixed when 1.05 comes out.

TianpeiLee, 2024-05-31:
Same as  #44

closed 2024-05-31 by TianpeiLee