openwch/arduino_core_ch32 #91: No objects Serial2, Serial3,... despite the existence of corresponding peripherals

michpro, 2024-05-11:
For chips having more than one U(S)ART port, the corresponding Serial2, Serial3,... objects should be created during compilation, but are not created.
Attempting to use such a non-existent object causes an obvious compilation error.
The problem lies in the design of the ‘WSerial.h’ file.

TianpeiLee, 2024-06-05:
When I try to change "Serial" to "serial2" it works fine, of course for "UART6" I did write it as "USART6"

[image]
[image]

maxint-rd, 2025-04-24:
FYI: for the CH32X033 I got dual UART working. Changing the definition in variant_xxx.h allows for selection of UART1, UART2 or both. See also issue #159. I hope to submit a updated PR sometime soon (after first doing some cleanup).
I expect the X035 to be similar.

medtty, 2026-10-28:
Only Serial is created by default to save RAM. Other ports work when declared in the sketch: `HardwareSerial Serial2(USART2)`; then `Serial2.begin(115200)`;. Closing as answered.

closed 2026-10-28 by medtty