openwch/arduino_core_ch32 #93: An interrupt-based implementation of HardwareSerial

michpro, 2024-05-13:
The currently available implementation of HardwareSerial is very primitive and is based on polling.
Especially when sending data, this causes unnecessary blocking of the program's execution of other tasks.
A DMA-based implementation, although it would relieve the CPU, will not be optimal for the variable amount of data transferred, hence I propose a traditional interrupt-based implementation.

maxint-rd, 2024-05-13:
The current implementation is not just primitive polling, it's also also missing some Arduino core functionality such as available() and peek().

For my own projects I made up a polling routine to fill the buffer used to [implement available() and peek()](https://github.com/openwch/arduino_core_ch32/compare/main...maxint-rd:arduino_core_ch32:Enable-Serial.available()-and-Serial.peek()). 

@tanakamasayuki proposed PR #79 as a simple improvement to implement available() and also suggested using interrupts, illustrated with [some code](https://gist.github.com/tanakamasayuki/56251790ad3c06a9837895054fd2f15e).

+1 for a nice interrupt based implementation. I'm in favor of anything better, but currently prefer small code size, since I found use of the limited flash on the CH32V003 the most prevalent issue in my projects...

maxint-rd, 2024-09-12:
Just a moment ago i submitted PR #145, which implements interrupt driven serial1. It enables Serial.available(), Serial.peek() and improves Serial.read(). Tested mainly on CH32V003, but also confirmed to work on CH32X035. Would be nice if someone could test other CH32 chips.

medtty, 2026-11-07:
Fixed by https://github.com/openwch/arduino_core_ch32/pull/284 ,Serial now receives with an RX interrupt and a 64-byte buffer on every UART and series, so available() returns the number of bytes waiting and nothing is lost while the sketch is busy. Tested on V003, V006, X035, V103, V203 and V307 EVT boards; it will be in the next release

closed 2026-11-07 by medtty