openwch/arduino_core_ch32 #95: CH32X035 ADC and GPIO

taisirou, 2024-05-17:
PC3 pin is not recognized.
If I changed the 43rd line from PIN_A13 to 20, it was recognized.
variant_CH32X035G8U.h

   #define PC3               20

ADC does not work properly.
I don't know the solution.
Of course ADC is effective.

   #define ADC_MODULE_ENABLED

MostafaKhan1971, 2024-09-30:
> Note 3: The channel 3, channel 7, channel 11, channel 15 and I2C functions of the ADC are not applicable to
> products with a lot number with the penultimate bit 5 being 0.

Is it a note from any of the source files?

MostafaKhan1971, 2024-09-30:
> ADC is not working properly on some pins.
> PA3,PA7,PC3 I don't know the solution.

I have manually downloaded the repository instead of downloading 1.0.4 through the board manager. PA3 and PA7 is working fine for me but PC3 is not. In some variants of X035 chips, PC3 can be used as a reset pin. Not sure if something about that got mixed up in the core.

MostafaKhan1971, 2024-09-30:
I have been struggling with this same issue for past couple of hours and could not find any obvious issues with library files.
When using the pin PC3/PIN_A13/A13 (all refer to the same GPIO Pin) in digital write or read or analog read function, the result I get is to and from pin PA1. But if I use the pin number 20 for digital read and write then it works fine and sends or receives signals to and from PC3. But analog read always reads 0 if pin number 20 is used. FYI, I am working with Arduino.

maxint-rd, 2024-10-18:
@MostafaKhan1971 - I recently received some CH32X033 chips. I checked the lot number and touch luck... I got the 0-series. I got the TSSOP20 version, so no A7 pin present, but it does have pins for A3 and I2C. 

Today I was testing more ADC and from what I saw so far I'm afraid they also lack working PADC_VREF. That internal reference voltage pin is on channel 15, which is unfortunately also listed in that note. So reading VCC, or converting any other ADC reading to Voltage seems impossible without an external VREF. 
(Oh, on that topic: it seems that due to a difference in capitalization, the Arduino constant VREF is not defined....)

When testing the use of PADC_VREF in analogRead(), it seemed to return the reading of A0. If that pin is not connected it appears to return a reflection of the voltage on other analog pins that are connected such as A1. The reading is a bit narrower on both top and bottom, but otherwise follows any voltage present on that other pin.

closed 2025-05-14 by yangliangquan