openwch/arduino_core_ch32 #101: Cant install core in Arduino

gonzzzales, 2024-05-29:
If i try to install this core through Board Manager in Arduino IDE 2.x i have this error

Failed to install platform: 'CH32 MCU EVT Boards:1.0.4'. 2 UNKNOWN: Get "https://codeload.github.com/openwch/risc-none-embed-gcc/zip/refs/tags/8.2.0": tls: failed to verify certificate: x509: certificate is valid for dns.google, dns.google.com, *.dns.google.com, 8888.google, dns64.dns.google, not codeload.github.com

How to install it manually?

TianpeiLee, 2024-05-30:
This looks a bit like a network issue,If you need to manually install, you can try doing the following（Windows example）:

1. First, make sure that your IDE version is 2. x
2. Find the installation path for your local library files, which usually looks like this：C:\Users\TempersLee\AppData\Local\Arduino15\packages
3. Recursively create a new folder .........../WCH/hardware/ch32v/1.0.4 under  "package"
4. Download the repository file to this directory
[image]
5. Similarly, recursively creating a new folder ............/WCH/tools/openocd/1.0.0  &  ........../WCH\tools\riscv-none-embed-gcc\8.2.0
6. Then download and decompress the openocd and gcc toolchains separately to the corresponding directories
7. 
openocd:https://codeload.github.com/openwch/openocd_wch/zip/refs/tags/1.0.0
[image]
gcc:https://codeload.github.com/openwch/risc-none-embed-gcc/zip/refs/tags/8.2.0
[image]

BTW: Please ensure that you have correctly installed the driver for Linke and that your estimated version of Linke is the latest

gonzzzales, 2024-05-30:
thanks a lot for manual. I will try it

TianpeiLee, 2024-05-31:
Good luck for that.

closed 2024-05-31 by TianpeiLee