openwch/arduino_core_ch32 #106: STOP mode for CH32V003?

radwouters, 2024-06-02:
In the datasheet of the CH32V003, section 2.3, it says that sleep is supported. When looking at ch32v00x_pwr.h, however, only the _standby_ [is implemented](https://github.com/openwch/arduino_core_ch32/blob/96e1e59f0ef19c4e045b799c668b553f52526d2c/system/CH32V00x/SRC/Peripheral/inc/ch32v00x_pwr.h#L62). Is there a reason for this? Would it be possible to use the same implementation as other chips and create a PR for this?

maxgerhardt, 2024-06-29:
See https://www.wch-ic.com/downloads/CH32V003RM_PDF.html and https://www.wch-ic.com/downloads/CH32V003RM_PDF.html

[image]
[image]

So, "sleep" mode in the RISC-V core is just entered with the wait for interrupt (WFI) or wait for event (WFE) instruction, these are already exposed via other file, e.g. `core_riscv.h`,

```cpp
// set SLEEPDEEP=0 to not enter into standby mode, only sleep mode
PFIC->SCTLR &= ~(1u << 2u);
// goodnight
__WFI();
```

radwouters, 2024-06-29:
I think I understand what is happening here, but I am not sure how you got to it. Is there a place where I can see this translation from register to c code? For example how PWR->CTLR is PDDS?

It would also be possible to make a PR for this, right? So add a PWR_EnterSLEEPMode-method below [here](https://github.com/openwch/arduino_core_ch32/blob/96e1e59f0ef19c4e045b799c668b553f52526d2c/system/CH32V00x/SRC/Peripheral/src/ch32v00x_pwr.c#L170)?

Please let me know if that would help the project :) Otherwise I'll just stay away from it!

maxint-rd, 2024-08-15:
FYI: I'm currently working on some easy functions in a fork of Adafruit's SleepyDog. I have something working (using the functions you mentioned) and intend to publish my work-in-progress in my GitHub account  some time soon. To be continued...

maxint-rd, 2024-10-01:
Hello @radwouters , I published my fork of [Adafruit_SleepyDog](https://github.com/maxint-rd/Adafruit_SleepyDog) that supports the CH32V003. 
It includes simple examples for using the watchdog and for using sleep. For longer sleeps I implemented repeated naps. Although not highly accurate, the millis counter is updated to reflect the time spent sleeping.
Would be nice if you coukd give it a test and let me know your findings.