openwch/arduino_core_ch32 #118: About serial port verification format

4061N, 2024-06-09:
Arduino 8E1 format is 8 bits of data, one even check, one stop, a total of 10 bits of data.

The traditional CH32 and STM32 are 7-bit data one-bit check, so there is one less.
To solve this problem, the databits needs to be incremented by one when there is a check bit.

For example, this is the code for HardwareSerial::begin in the stm32duino project：
`void HardwareSerial::begin(unsigned long baud, byte config)
{
  uint32_t databits = 0;
  uint32_t stopbits = 0;
  uint32_t parity = 0;

  _baud = baud;
  _config = config;

  // Manage databits
  switch (config & 0x07) {
    case 0x02:
      databits = 6;
      break;
    case 0x04:
      databits = 7;
      break;
    case 0x06:
      databits = 8;
      break;
    default:
      databits = 0;
      break;
  }

  if ((config & 0x30) == 0x30) {
    parity = UART_PARITY_ODD;
    databits++;
  } else if ((config & 0x20) == 0x20) {
    parity = UART_PARITY_EVEN;
    databits++;
  } else {
    parity = UART_PARITY_NONE;
  }

  if ((config & 0x08) == 0x08) {
    stopbits = UART_STOPBITS_2;
  } else {
    stopbits = UART_STOPBITS_1;
  }

  switch (databits) {
#ifdef UART_WORDLENGTH_7B
    case 7:
      databits = UART_WORDLENGTH_7B;
      break;
#endif
    case 8:
      databits = UART_WORDLENGTH_8B;
      break;
    case 9:
      databits = UART_WORDLENGTH_9B;
      break;
    default:
    case 0:
      Error_Handler();
      break;
  }

  uart_init(&_serial, (uint32_t)baud, databits, parity, stopbits);
  enableHalfDuplexRx();
  uart_attach_rx_callback(&_serial, _rx_complete_irq);
}`

It can be see that the databits is incremented when there are check bits. I think this part of CH32 also needs to be fixed.

maxint-rd, 2024-06-09:
Just to be complete; the Arduino default is N81: 
>"The default is 8 data bits, no parity, one stop bit."
See also the [Arduino documentation](https://www.arduino.cc/reference/en/language/functions/communication/serial/begin/) for Serial.begin(). In my opinion this CH32 core should aim for maximal compatibility and try to follow the Arduino core as close as possible.

4061N, 2024-06-11:
> Just to be complete; the Arduino default is N81:
> 
> > "The default is 8 data bits, no parity, one stop bit."
> 
> See also the [Arduino documentation for Serial.begin()](https://www.arduino.cc/reference/en/language/functions/communication/serial/begin/). In my opinion this CH32 core should aim for maximal compatibility and try to follow the Arduino core as close as possible.
> 
> Perhaps more importantly with regards to compatibility: this core still misses some core functionality for Serial such as Serial.available() and Serial.peek(). For my own project I made some cumbersome fixes. A proper interrupt driven implementation is still on my wish-list. See also issue #93 and closed PR #79.
Try use 8E1 to send data, and the serial port software will not read out the correct 8E1, because CH32 and STM32 calculate the check bit into the data, which makes 8E1 equal to the common Arduino 7E1. In the Ardino adaptation of STM32, they have solved this problem, which is to add one to the databits when it is judged to be parity, but I found that in the CH32 code, it did not do so, which caused some Arduino programs that rely on serial ports to have errors, which I think should be fixed

closed 2026-10-28 by medtty