openwch/arduino_core_ch32 #119: request for OPA example

garudaonekh, 2024-06-13:
Hi,
I would like to use CH32V003 OPA(Op-Amps) to connect to PH sensor. But I have no idea how to set this in Arduino.

I found a git which use this  [](https://github.com/MoonFox2006/CH32V003_OPA)

Thanks;

maxint-rd, 2024-06-13:
I had a brief look into that git. From what I saw it appears the author set up his own Arduino-like environment in MounRiver. 

The file you're looking for is [/User/main.cpp](https://github.com/MoonFox2006/CH32V003_OPA/blob/main/User/main.cpp). It contains the familiar setup() and loop() functions. In setup() you can see how ADC is prepared to use OPA and initialize the interrupt handler that calculates the average ADC reading. The loop() function is used to display that average on a TM1637 display.

A first attempt to make an OPA example could be based on that code. Just rename that main.cpp to adc_opa.ino, replace the TM1637 code by Serial.println() and see what else needs to change to get it working. 

To make this into a proper example once something is working, all register manipulation should be replaced by calls to functions provided by this core. I don't know to what extend this core already supports OPA and I'm currently not in the position to spend time on this. Hopefully my analysis was helpful to get you started. Good luck!

garudaonekh, 2024-06-14:
> I had a brief look into that git. From what I saw it appears the author set up his own Arduino-like environment in MounRiver.
>
> The file you're looking for is [...]

Thanks. Yes, the hardest part is to manipulate the register values. I will try it

maxint-rd, 2024-10-13:
Some more reading about using OPA [here](https://www.eevblog.com/forum/microcontrollers/ch32v003-ch32x035-how-to-configure-the-opa/), with explanation of pin usage and copy of code from the WCH example.

closed 2024-10-15 by garudaonekh