openwch/arduino_core_ch32 #122: analogWrite for ch32x035 not workign

arwidcool, 2024-06-17:
As the title says, no support for analogWrite();

What needs to be done in order for it to work?

Has anyone figured it out?

maxint-rd, 2024-06-18:
According to issue #31 it should be working with the latest sources, just not with the latest release. Have you tried latest commits?

arwidcool, 2024-06-19:
It seems like any of the PWM pins using TIM1 causes either no result or actually crashes the MCU.

Using PB0 creates junk from serial almost immedatley

arwidcool, 2024-06-19:
Update on Pins confirmed NOT working:

PA7, PB1, PC19 (This is SWCLK), PB12,PB3, PB10??(Serial TX) and PB0 is completely messed up causes serial junk when used.

[image]

mrangen, 2024-08-04:
I'm struggling to get PWM working as well.

I downloaded the latest version of this repo, unzipped it and used its contents to replace everything in this folder:
C:\Users\myusername\AppData\Local\Arduino15\packages\WCH\hardware\ch32v\1.0.4

I still can't get this code to do anything but blink the LED like a digital output. I tried the pins you listed above as working but still no PWM, it's just switches on once it's past 50% duty cycle e.g. 0 to 2050 the LED is off, 2051 to 4095 the LED is on.

Have I missed a step?

[image]

arwidcool, 2024-08-05:
@mrangen you seem to be using Arduino IDE. Get it running on PlatformIO @ https://github.com/Community-PIO-CH32V/platform-ch32v

That is what I am using and the mentioned pins work with it. Probably using a different version of the core and maybe buildtools.

mrangen, 2024-08-06:
@arwidcool thanks, got PWM working now that I'm in PlatformIO, cheers!

maxint-rd, 2024-08-07:
Currently I only have some models of the CH32V003 to experiment with and have no serious complaints when using the Arduino IDE v2.3.2 . If I had a v035 with such issue, I would first try the same thing out on the v003. (I just tested a v003 with recent code and had working PWM working fine with frequencies as low as 1Hz and as high as 2MHz)

maxint-rd, 2024-08-29:
@zwieblum - hmm... are you sure? digitalWrite() has worked for me in every single release I tried and all my CH32V003 projects use it in various ways. Not having digitalWrite() would make the MCU kind of useless.
Are you sure your test environment is setup correctly? Did you try other basic example? What did work and what did not?

zwieblum, 2024-08-29:
Quite sure. I did not get any pin toggling that is connected to TIM1, e.g. PD0.  ch32c003fun pwm timer1 example works without a problem for all 4 channels. I did not try pins connected to timer2 in arduinoIDE. 
To clearify: I use ch32v003fun for development, but I would be happy to use the ArduinoIDE for the sutdents. I do not use platformIO.

maxint-rd, 2024-08-29:
Okey, so perhaps you could try analogWrite() instead of digitalWrite().  
For analogWrite() to work you need:
 1. the most recent source code from the master branch, 
 2. the Hardware Timer needs to be enabled in the applicable header file (don't recall the exact name since I modified the board.txt to do it from the the menu)
 3. use a supported pin. In my most recent tests with a SOP16 CH32V003 I found this: 
      PWM SOP16 tested working: PA1/PC0/PC3/PC4 - 
      PWM not working: PC1/PC2/PC6/PC7 (only on/off); PD7=/RST; PA2: always off
When a pin is not supported analogWrite() will switch on above 50% and off below 50%.

For my fan controller project I did not do any PWM register configuration directly myself, I only used these:
```
  pinMode(fan, OUTPUT);
  analogWriteFrequency(100);      // low freq. is much better for driving fan motor
  analogWrite(fan, nSpeed);
```

zwieblum, 2024-08-29:
Oops, just saw the copy&pase error, too :)

With or without `analogWriteFrequency(100);  I get no PWM on any pin.
Timer module is enabled by default (it's in the chip definition, TIM_MODULE_ENABLED). Each call to analogWrite() calls pwm_start() which end up reinitialising the timer, but in a way that it is not started. 

Did you test on ArduinoIDE or PlatformIO?

maxint-rd, 2024-08-29:
> Did you test on ArduinoIDE or PlatformIO?

I'm using Arduino IDE 2.3.2 on Windows 11. I first installed core v1.4, then copied newer files over from the master branch and made some changes to boards.txt to let me enable modules using the IDE menu. 
In my first test I used the Fade example (without setFrequency), then modified it to do some more detailed tests. I measured the default PWM frequency to be 1kHz which was too whiny for a MOSFET driven fan.

This is the bare_minumum example that I adjusted for basic testing/debugging:
```
int fan = PA1;    // PA1: the PWM pin the LED is attached to 

void setup() {
  // put your setup code here, to run once:
  pinMode(fan, OUTPUT);
  analogWriteFrequency(100);      // low freq. is much better for fan motor; too high causes fan to remain off, default speed 1000 gives whining noise
}

void loop() {
  static int nSpeed=0;
  // put your main code here, to run repeatedly:
  analogWrite(fan, nSpeed); // 8812-4732=4080 bytes for analogwrite!
  if((nSpeed+=10)>4096) nSpeed=0;
  delay(100);
}
```

I just now tested it to be working (sill). Using the debugger with a breakpoint on the analogWrite() statement, I saw it calling pwm_start() on line 170 of wiring_analog.c. The call to set_pin_configured() on line 167 was done only once, in the first iteration of loop().

zwieblum, 2024-08-29:
"1.4" or "1.0.4"? Arduino Boardmanager installs 1.0.4, which is identical to the git version.
The additional boardmanager URL is https://github.com/openwch/board_manager_files/raw/main/package_ch32v_index.json

The example you gave does not work here.

maxint-rd, 2024-08-29:
> "1.4" or "1.0.4"?

Sorry. Meant 1.0.4. indeed, which is the latest release via the board manager indeed. (edited just now)
On top of that release you also need all the latest commits related to the hardware timer. Based on your earlier posts I think you already have those, but it still doesn't work for you. If I were you I'd probably try to trace things in the debugger. 

For your information, this is the size of the compiled bare_minimum example above, with hardware timer module enabled and optimized for debugging, with debug symbols:
```
Sketch uses 9860 bytes (60%) of program storage space. Maximum is 16384 bytes.
Global variables use 436 bytes (21%) of dynamic memory, leaving 1612 bytes for local variables. Maximum is 2048 bytes.
```

If it's much less for you, you might be missing some core components.

zwieblum, 2024-08-29:
I'm on Linux, there is no permanent cache (not even something that survives restarting the IDE) :)

```
Der Sketch verwendet 10700 Bytes (65%) des Programmspeicherplatzes. Das Maximum sind 16384 Bytes.
Globale Variablen verwenden 592 Bytes (28%) des dynamischen Speichers, 1456 Bytes für lokale Variablen verbleiben. Das Maximum sind 2048 Bytes.
```
... so size is simillar to yours.

Just for fun, the "same" program on c32v003fun works and is a bit smaller:
```
Memory region         Used Size  Region Size  %age Used
           FLASH:        2076 B        16 KB     12.67%
             RAM:          0 GB         2 KB      0.00%
```

Anyway, as the ch32v003fun code works on my hardware and the arduino code not I assume it's safe to say the arduino code is (still) broken.

maxint-rd, 2024-08-30:
The Windows IDE apparently keeps much more things alive. Not only compiled core code is stored in a build folder, but the IDE also stores IDE window settings such as selected board and clock speed.

Binaries produced by ch32v003fun are definitely quite a bit smaller, at the effort of lower level coding. Personally I'm just a maker experimenting with different MCU's and prefer the platform independent nature of Arduino code. For the CH32V003 those much larger binaries can be quite challenging. Having smaller binaries, e.g. by getting LTO working is still on my wish list...
BTW. Alexander Mandera made the [arduino-wch32v003](https://github.com/AlexanderMandera/arduino-wch32v003) core based on ch32v003fun to bring smaller binaries to the Arduino IDE, but unfortunately that development doesn't show much progress and still lacks much support for peripherals.

I think the least we can conclude that this Arduino core seems broken for Linux where it concerns PWM. Unfortunately this core is depending on the voluntary efforts of @TianpeiLee and a small group of contributors. Too bad I couldn't be of more help...

zwieblum, 2024-09-07:
Ok, aperently arduino boardmanager is to balme for this whole issue: it insists on installing 1.0.4 which does not have a working code for analogWrite() - but master has (including SPI ...), so you need to do some surgery to get it working:

```
cd ~/.arduino15/packages/WCH/hardware/ch32v
rm -R 1.0.4
git clone --depth=1 https://github.com/openwch/arduino_core_ch32 1.0.4
```

Now analogWrite works as intended for all CH32V :)

maxint-rd, 2024-09-07:
Good to read you got it working. The current master branch has quite a few useful improvements, including PWM. (BTW. I only have the CH32V003, so I couldn't test my own contributions (eg. I2C slave) on anything else. )

> 1.0.4 indeed; which is the latest release via the board manager indeed [...] On top of that release you also need all the latest commits related to the hardware timer.

maxint-rd, 2024-10-12:
FYI: After getting some CH32X033F8P6 (TSSOP20) chips I just finished testing PWM. Only one pin had PWM working: PA6/T3C1. Perhaps no coincident: PA6 is the only working pin on the [X035 PWM list](https://github.com/openwch/arduino_core_ch32/issues/122#issuecomment-2179519665) by @arwidcool  that is also on the X033 F8P6.

closed 2025-05-14 by yangliangquan