openwch/arduino_core_ch32 #123: Hardware reset pin doesn't work on CH32X033 or CH32X035

zane-c, 2024-06-19:
Sorry if this is the wrong repo to be asking this question. I have a few boards with the CH32 chips. My reset circuit works fine for the v003 chips, but doesn't work for the X03x chips. I am wondering if this is an issue with the defaults being set in the Arduino core for these chips or something else completely.

Both appear to have identical reset circuits in the datasheets. But I did find it odd that one is RST and one is NRST. Shouldn't RST be active high?
- [Ch32x reset circuit](https://github.com/openwch/arduino_core_ch32/assets/25088418/4f0b1569-922b-4262-867f-7de9872d81b3)
- [CH32v00x reset circuit](https://github.com/openwch/arduino_core_ch32/assets/25088418/0e31e8e4-551e-4c06-a673-1a4ab81d3f4a)

My circuits for reference:
- [CH32x schematic](https://github.com/openwch/arduino_core_ch32/assets/25088418/776f1ebf-8714-4c14-ace1-57c377181500)
- [CH32v00x schematic](https://github.com/openwch/arduino_core_ch32/assets/25088418/18e34154-c22c-4e09-9460-9823a46b0747)

Sorry again for the noob question. I am not an embedded engineer.

maxint-rd, 2024-06-19:
At first look your circuit seems fine. Reset is usually active low indeed. Those NRST/RST labels may cause some confusion.

My one case where RST was not working was when I used the particular pin for something else.  I see in your schematic that for the X035, the RST pin uses PC3. Are you sure your sketch doesn't use PC3 for something? Could it be that PC3 is already initialized for something else by the core? Unfortunately I don't have any X035 to test with...

zane-c, 2024-06-19:
Thanks @maxint-rd for taking a look and for the suggestion. I can confirm my sketches don't have any handling of the PC3 pin. I will keep poking through the core's code to see if I can find anything that is initializing that to a non-default.

maxint-rd, 2024-10-10:
And.. Did you find something?

I finally got a bunch of CH32X033F8P6 (TSSOP20) chips and soldered a couple to breakout boards. I can confirm that for my X033 pin 4 RST doesn't reset the processor either. Using F12 in the Link Utility does successfully reset the processor.

Once more I reviewed the datasheet. For pin 4 it lists PB7 as main function after reset with default alternate function RST/T1C2N/O2P2/RTS3. Comparing that to the CH32V003 datasheet lists pin 4 having PD7  as main function after reset with default alternate function NRST/T2CH4/OPP1. No help there...
(For completion: the datasheet lists RST at pins 8/PC3, 4/PC3 and 7/PA21 for the X035 QSOP28, TSSOP20, LQFP64M and LQFP48. No RST for the QFN28 and QFN20?)

The [reference manual](https://www.wch-ic.com/downloads/CH32X035RM_PDF.html) may shed some more light. On page 246 It states for User Option Bytes [4:3] RST_MODE this:
> External reset pin RST enable:
> 00: RST multiplexing enabled;
> 11: Multiplexing function off, PA21/PC3/PB7 for GPIO function.
> Note: For reset pins in different packages, refer to the datasheet for pin descriptions. For example, PA21 can be multiplexed as an external reset pin in 64-pin R8T6 and 48-pin C8T6 packages.

Reading those option bytes with the Link Utility gave these values:

|Adress |  Bit [31:24] | [23:16] | [15:8] |  [7:0] |
|--------|-------------|---------|-------|-------|
| 0x1FFFF800 |  nUSER: A5 | USER: 5A | nRDPR: 1F | RDPR: E0 |
| 0x1FFFF804 | nData1: 00 | Data1: FF | nData0: 00 | Data0: FF |
| 0x1FFFF808 | nWRPR1: FF | WRPR1: 00 | nWRPR0: FF | WRPR0: 00 | 
| 0x1FFFF80C | nWRPR3: FF | WRPR3: 00 | nWRPR2: FF | WRPR2: 00 |

RDPR value 0xE0 is binary 1110 0000, showing value 00b for RST_MODE bits [4:3]. This should mean RST multiplexing enabled. However, we both see that while multiplexing might be enabled, the pin doesn't work as /RST. No help still...

FYI: The V003 reference manual give the same option byte, with these reset modes:
> PD7 multiplexed as external pin reset.
> 00: Ignoring pin states within 128us after turning on the multiplexing function.
> 01: Ignoring pin states within 1ms after turning on the multiplexing function.
> 10: Ignoring pin states within 12ms after turning on the multiplexing function.
> 11: Multiplexing function off, PD7 for I/O function.

BTW. Stefan Wagner made a [CH32X033 development board](https://oshwlab.com/wagiminator/ch32x033f8p6-development-board). It has a reset button that ties PB7 to GND with a 100n capacitor parallel to the button. Nothing special as a circuit I believe.

In short: I didn't find anything... Still looking and hoping for some clues...

maxint-rd, 2024-10-12:
@zane-c - Got hardware reset pin working as reset on my CH32X033F8P6. See previous post...

zane-c, 2025-02-15:
Thanks @maxint-rd for solving this one. I finally had some time to revisit this and you made it easy for me :)

So I guess disabled RST is the default for this chip so this technically is WAI; I'll close this issue.

Would have been nice if this was mentioned in the datasheet but I guess I should have looked closer at the configuration options in the link utility.

Anyway, thanks again!

closed 2025-02-15 by zane-c