openwch/arduino_core_ch32 #124: i2c_scanner.ino build regression after PR #108

lyusupov, 2024-06-19:
### Build platform
Linux

### Core version
current 'main' branch

### Target
CH32V30x_EVT

### To reproduce

```
$ export BOARD=WCH:ch32v:CH32V30x_EVT
$ arduino-cli compile -v --build-path=/tmp/arduino -b "$BOARD" libraries/Wire/examples/i2c_scanner/

<... skipped ...>

/home/codespace/.arduino15/packages/WCH/tools/riscv-none-embed-gcc/8.2.0/bin/riscv-none-embed-gcc -DSYSCLK_FREQ_144MHz_HSI=144000000 -march=rv32imafcxw -mabi=ilp32f -msmall-data-limit=8 -msave-restore -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -fno-common -c -Os -DNDEBUG -w -std=gnu99 -MMD -I/home/codespace/.arduino15/packages/WCH/hardware/ch32v/1.0.4/libraries/Wire/examples/i2c_scanner -I/home/codespace/.arduino15/packages/WCH/hardware/ch32v/1.0.4/cores/arduino -I/home/codespace/.arduino15/packages/WCH/hardware/ch32v/1.0.4/cores/arduino/avr/ -I/home/codespace/.arduino15/packages/WCH/hardware/ch32v/1.0.4/cores/arduino/ch32/ -I/home/codespace/.arduino15/packages/WCH/hardware/ch32v/1.0.4/cores/arduino/ch32/lib/ -I/home/codespace/.arduino15/packages/WCH/hardware/ch32v/1.0.4/system/CH32V30x/USER/ -I/home/codespace/.arduino15/packages/WCH/hardware/ch32v/1.0.4/system/CH32V30x/SRC/Core/ -I/home/codespace/.arduino15/packages/WCH/hardware/ch32v/1.0.4/system/CH32V30x/SRC/Debug/ -I/home/codespace/.arduino15/packages/WCH/hardware/ch32v/1.0.4/system/CH32V30x/SRC/Startup/ -I/home/codespace/.arduino15/packages/WCH/hardware/ch32v/1.0.4/system/CH32V30x/SRC/Peripheral/inc/ -I/home/codespace/.arduino15/packages/WCH/hardware/ch32v/1.0.4/system/CH32V30x/SRC/Peripheral/src/ -DARDUINO_ARCH_CH32 -DCH32V30x -DARDUINO=10607 -DCH32V30x_C "-DVARIANT_H=\"variant_CH32V307VCT6.h\"" -I/home/codespace/.arduino15/packages/WCH/hardware/ch32v/1.0.4/cores/arduino -I/home/codespace/.arduino15/packages/WCH/hardware/ch32v/1.0.4/variants/CH32V30x/CH32V307VCT6 -I/home/codespace/.arduino15/packages/WCH/hardware/ch32v/1.0.4/libraries/Wire/src -o /tmp/arduino/libraries/Wire/utility/twi.c.o -c /home/codespace/.arduino15/packages/WCH/hardware/ch32v/1.0.4/libraries/Wire/src/utility/twi.c
/home/codespace/.arduino15/packages/WCH/hardware/ch32v/1.0.4/libraries/Wire/src/utility/twi.c: In function 'I2C2_EV_IRQHandler':
/home/codespace/.arduino15/packages/WCH/hardware/ch32v/1.0.4/libraries/Wire/src/utility/twi.c:686:34: error: 'handle' undeclared (first use in this function)
    i2c_slave_process(get_i2c_obj(handle));  // process I2C transmissions, for now only events, not errors
                                  ^~~~~~
/home/codespace/.arduino15/packages/WCH/hardware/ch32v/1.0.4/libraries/Wire/src/utility/twi.c:686:34: note: each undeclared identifier is reported only once for each function it appears in

```

### Root cause

The PR #108 had set

```
 #define OPT_I2C_SLAVE 1
```
in twi.h: 
https://github.com/openwch/arduino_core_ch32/blob/main/libraries/Wire/src/utility/twi.h#L53-L56

This change works for CH32V00x_EVT target build but it **breaks CH32V30x_EVT one** !!!

maxint-rd, 2024-06-19:
Allright, I think I found the cause.
For the V003, the code seems only compiled for I2C1_BASE. Appearantly V30x also has I2C2_BASE defined. That undeclared "handle" is declared a few lines higher. For I2C1_BASE this code is used:
```
void I2C1_EV_IRQHandler(void)
{
#if OPT_I2C_SLAVE
   I2C_HandleTypeDef *handle = i2c_handles[I2C1_INDEX];  // MMOLE: was commented
   // MMOLE: I2C1_EV_IRQHandler is the event handler, handle is an I2C_HandleTypeDef struct containing parameters and pointer to the registers
   static int _nCounterEV1=1;
   _nCounterEV1++;
   i2c_slave_process(get_i2c_obj(handle));	// process I2C transmissions, for now only events, not errors
#endif
}
```

Guess what... for I2C2_BASE the code is almost identical:
```
void I2C2_EV_IRQHandler(void)
{
#if OPT_I2C_SLAVE
   //I2C_HandleTypeDef *handle = i2c_handles[I2C2_INDEX];  // MMOLE: was commented
   // MMOLE: I2C2_EV_IRQHandler is the event handler, handle is an I2C_HandleTypeDef struct containing parameters and pointer to the registers
   static int _nCounterEV2=1;
   _nCounterEV2++;
   i2c_slave_process(get_i2c_obj(handle));	// process I2C transmissions, for now only events, not errors
   // MMOLLE: tested only using I2C1
#endif
}
```

As you can read in that final comment,  I only tested using I2C1. Never compiled for V30x. When looking at the definition of "handle", you can see it is still commented for that I2C2_BASE. It only needs to be uncommented to make the code compilable. Just remove the first slashes from this line:
`//I2C_HandleTypeDef *handle = i2c_handles[I2C2_INDEX];  // MMOLE: was commented`
 I just did it for my local copy and used the i2cMinimalSlave example to see if it compiles. Before if gave the same error you quoted. Now it properly compiles for the CH32V30x board. Mind you, I still don't have any V30x hardware to test this. 

If you don't need I2C slave functionality I  suggest you to switch the slave option off. Then all related code should be disabled and make the code compile as well:
` #define OPT_I2C_SLAVE 0`

@lyusupov - If I can find some more time I will make the fix in a new PR. Perhaps in the mean time you can make your own local change and see if that i2cMinimalSlave example actually works for the CH32V30x? Thanks for your testing and good luck compiling!

lyusupov, 2024-06-20:
I can confirm V30x build fix after this commit https://github.com/maxint-rd/arduino_core_ch32/commit/b9fa7d8f7503c3e449b15d86bbcaaff0d2ea2724

Thank you!

closed 2024-06-20 by lyusupov