openwch/arduino_core_ch32 #130: Can't find wch board in board manages, arduino 2.3.2

kylongmu, 2024-07-03:
Add this in the Preferences：
https://github.com/openwch/board_manager_files/raw/main/package_ch32v_index.json
But can't find whc board in the board manages, my arduino version is 2.3.2

vimalb, 2024-07-03:
Might be related to the issue [reported here](https://github.com/openwch/arduino_core_ch32/issues/117) - try searching for "CH32" instead of "WCH" to find the this core.

Also be aware of [this issue](https://github.com/openwch/arduino_core_ch32/issues/44) that causes 1.0.4 to fail to install on all versions of Windows - there are some workarounds in the thread including running as admin or manually tweaking/installing the core

kylongmu, 2024-07-05:
> Might be related to the issue [reported here](https://github.com/openwch/arduino_core_ch32/issues/117) - try searching for "CH32" instead of "WCH" to find the this core.
>
> Also be aware of [this [...]

Can't find CH32 also, and tried running arduino as admin, but has nothing happen.

kylongmu, 2024-07-07:
https://github.com/openwch/board_manager_files/raw/main/package_ch32v_index.json
is this file losted? can't open it.

maxint-rd, 2024-07-07:
If I open your first link, it can open, but it's redirected to this:
https://raw.githubusercontent.com/openwch/board_manager_files/main/package_ch32v_index.json

vimalb, 2024-07-11:
The redirects shouldn't matter - Arduino IDE should follow them automatically and resolve to the final content.

As an experiment, I did a clean install of Arduino IDE 2.3.2 on Windows 10

I then added the board manager URL per the installation instructions:
[image]

After that, I was able to search for "CH" and see the "CH32 MCU EVT Boards" core come up:
[image]

I'm quite convinced this core is published correctly and Arduino IDE 2.3.2 should be able to find and install it. 

One more thing to check - is this your only board manager URL, or do you have multiple?
If you have multiple, are they all separated by newlines (not commas or anything like that)?
Have you made sure there aren't any hidden whitespace or other characters in the URL when you copied-and-pasted it?

kylongmu, 2024-07-11:
> The redirects shouldn't matter - Arduino IDE should follow them automatically and resolve to the final content.
>
> As an experiment, I did a clean install of Arduino IDE 2.3.2 on Windows 10
>
> I then [...]

My OS is win11, do a clean install arduino 2.3.2 get this error:
[image]

TianpeiLee, 2024-07-12:
> Add this in the Preferences： https://github.com/openwch/board_manager_files/raw/main/package_ch32v_index.json But can't find whc board in the board manages, my arduino version is 2.3.2

[image]

No matter how I test, everything is normal

closed 2025-05-06 by yangliangquan