openwch/arduino_core_ch32 #134: Programming without LinkE?

thor171, 2024-07-05:
Hi,
I'm new to the CH32x035. I made a pcb for testing purposes and I'm wondering, how to program it using it's USB interface. The datasheet says: "Built-in 3328B System memory (Bootloader) for storing the system bootloader (Factory-cured bootloader)." Does "Factory-cured" mean, there is already a bootloader flashed by default?
If that's the case, is it possible to flash the device without the need for a LinkE?
Anyways, a little bit more documentation on the programming process in general would be great, I couldn't find much.

maxgerhardt, 2024-07-08:
Exactly, the chip has a USB bootloader. Some good info can e.g. be found in https://github.com/wagiminator/CH32X035-USB-PD-Tester?tab=readme-ov-file#compiling-and-uploading-the-firmware-linux.

Specifically: Bootloader must be entered by pulling the "UDP" / PC17 HIGH at boot, then releasing it, plus the hardware connections you need for the USB plug, etc. The tool [wchisp](https://github.com/ch32-rs/wchisp) can flash it then. As can the official WCHISPTool.

vimalb, 2024-07-08:
I think we're all anxious for another release to be cut - there are some awesome new features including tinyusb and wchisp integration: 
https://github.com/openwch/arduino_core_ch32/pull/129

You can test them out main branch if you'd like - credit to @hathach for submitting relevant MRs