openwch/arduino_core_ch32 #135: non-portable attachInterrupt() definition in ~/.arduino15/packages/WCH/hardware/ch32v/1.0.4/cores/arduino/WInterrupts.h

polypagan, 2024-07-16:
attachInterrupt(pin,GPIOMode_TypeDef io_mode, void (*callback)(void), EXTIMode_TypeDef it_mode, EXTITrigger_TypeDef trigger_mode);

throws errors.

attachInterrupt(digitalPinToInterrupt(pin), ISR, mode);

is used by every other Arduino package I've used/looked at. I assume this variation is required by features of CH32V parts. There must be some workaround that I am not smart enough to work out.

If I were calling attachInterrupt() directly from my own code, I could make this conditional. That approach doesn't work when attachInterrupt() is called by library code.

lyusupov, 2024-07-17:
duplicates https://github.com/openwch/arduino_core_ch32/issues/15

polypagan, 2024-07-17:
My apologies for duplicate. I did search, didn't find. 

I will close this, duplicate is also closed with refusal to correct, making this core unusable for many applications.

closed 2025-05-06 by yangliangquan