openwch/arduino_core_ch32 #139: Servo library for CH32V003

vanminh0910, 2024-09-03:
I am looking for some example code to control RC servo (like mini Servo SG90s) using CH32V003. The Arduino default Servo library is not compatiblee with CH32V003. Can anyone point me some references? Thanks.

maxint-rd, 2024-09-04:
See issue #30 for some comments on this topic.

Now that hardware timer is supported perhaps it's easier to implement support for this.
I found this [ch32v003fun servo project](https://github.com/limingjie/CH32V003-Servo) that may be handy as a reference.

 Currently I don't have spare time to look into this, but having servo support is on my wishlist too.

vanminh0910, 2024-09-04:
@maxint-rd Thanks for your info. 

I already looked at the link with ch32v003fun servo project but can barely understand the code to integrate into my Arduino project. Let me play with analogWrite. I made it work before. If not working with servo, I need to wait more for a ready to use library since my low level coding is not good enough.

maxint-rd, 2024-09-04:
I'm also no low level coding expert, but I like using the CH32V003 for it's low price and want to use it for all it can do. My wish is to have a fully compatible Arduino core. With some effort I managed to implement some things I needed, such as I2C slave, interrupt driven serial, reading VCC and enabling the watchdog. I published most work in my Github and some was incorporated as PR. 

With regards to the hardware timer, I already managed to use PWM in my fan controller project and with the latest code from the master branch analogWrite() should work. As far as I understand using a servo requires setting the right frequency and a proper pulse duration. Most servo code I saw sets a timer to toggle a pin at certain pulse length. The default frequency is 1kHz, but that can be adjusted using analogWriteFrequency(). The PWM resolution is 12 bits (0-4096) and your experiment could show if changing PWM-duty with a fixed frequency,  or changing frequency with a fixed duty cycle leads to usable results. 

Maybe at later moment can join your experiments and if I find some more time I will have a closer look at having a proper library...

zwieblum, 2024-09-07:
As @maxint-rd wrote: use analogWriteResolution(16); and analogWriteFrequency(50); to set the PWM basicc parameters and analogWrite() in the range of 3276 ... 6553. This should result in 1ms .. 2ms pulse width at 50 Hz, which is a norm signal for RC servos.

svofski, 2026-08-19:
analogWrite workaround does work for me on CH32V003F4P6, tested on pin PD3.