openwch/arduino_core_ch32 #140: Static link of LibUSB can bypass brew requirement for Mac

DeqingSun, 2024-09-04:
Brew is not friendly to older Mac OS systems or newbie users. LibUSB can be statically linked inside the binary.

Here is how I did it.

```
ifneq (,$(findstring darwin,$(OS_NAME)))
	LIBS=/usr/local/Cellar/libusb/1.0.23/lib/libusb-1.0.a /usr/local/Cellar/libusb-compat/0.1.5_1/lib/libusb.a -ldl -Wl,-framework,IOKit -Wl,-framework,CoreFoundation
else
ifneq (,$(findstring cygwin,$(OS_NAME)))
	LIBS=-lusb-1.0 -Wl,--enable-stdcall-fixup CH375DLL.dll
else
	LIBS=-lusb-1.0
endif
endif
```

Here is a Github Action build example
https://github.com/DeqingSun/vnproch551/blob/master/.github/workflows/build_for_multiple_os.yml

zoobab, 2025-02-25:
I made static builds for minichlink tool, cross-compiled from linux to windows here:

https://github.com/zoobab/ch32v003fun/tree/minichlink-static

I am wondering if I could cross-compile to OSX with GCC with just copying the required headers file (IOKit and Foundations).

DeqingSun, 2025-06-03:
> I made static builds for minichlink tool, cross-compiled from linux to windows here:
> 
> https://github.com/zoobab/ch32v003fun/tree/minichlink-static
> 
> I am wondering if I could cross-compile to OSX with GCC with just copying the required headers file (IOKit and Foundations).

You can try Github action that gives you a free Mac virtual machine. My link is an example to compile binary in Linux and Mac