openwch/arduino_core_ch32 #141: Why CH32V003J4M6 analogRead() not working ???

mindonne, 2024-09-08:
I tried everything but couldn't ? 
analogRead(PA_2) ?
analogRead(PA2) ?
all all return 0 :((( 
but when i use it digitalWrite(PA2) this not output 3.3v !!!
plssss help meeeeeeee

maxint-rd, 2024-09-08:
In my fan controller project I have ADC working on the SOP8 CH32V003J4M6. It uses A0 (PA2) to read analog values. First some questions for you:
- Do you have the latest core? (release 1.0.4)?
- Did you update it with the most recent commits of the master branch?
- Is ADC enabled?
- Do you have proper clock speed selected?
- Have you tried analogRead(A0)  instead of using PA_2 and did you use pinMode(A0, INPUT) ?

Please also read issue #110 for much info about using ADC. It has an example demonstrating proper use of ADC. Good luck!

mindonne, 2024-09-08:
> Trong dự án quạt điều khiển của tôi, tôi có ADC hoạt động trên SOP8 CH32V003J4M6. Nó sử dụng A0 (PA2) để đọc các giá trị tương tự. Trước đó, có rất nhiều câu hỏi dành cho bạn:
> 
> * Bạn có phiên bản mới nhất của lõi không? (phiên bản 1.0.4)?
> * Bạn đã cập nhật nó với phiên bản cam kết mới nhất của nhánh chính chưa?
> * ADC có được bật không?
> * Bạn đã chọn chưa phù hợp với tốc độ đồng hồ?
> * Bạn đã thử analogRead(A0) thay vì sử dụng PA_2 và bạn chưa sử dụng pinMode(A0, INPUT)?
> 
> Vui lòng đọc thêm số [110](https://github.com/openwch/arduino_core_ch32/issues/110) để biết thêm thông tin về cách sử dụng ADC. Có ví dụ minh họa cách sử dụng ADC đúng cách. Chúc may mắn!

1. I use ARDUINO IDE 2.3.2
2. My code : 
void setup() {
  pinMode(A0, INPUT_ANALOG);
  Serial.begin(9600);
}

void loop() {
  Serial.println(analogRead(A0));
  delay(100);
}

Serial : 
16:02:54.067 -> 664
16:02:54.245 -> 653
16:02:54.473 -> 636
16:02:54.694 -> 670
16:02:54.856 -> 663
16:02:55.081 -> 663
16:02:55.264 -> 665
16:02:55.468 -> 664
16:02:55.672 -> 652
16:02:55.878 -> 631

BUT when i connet điều khiển của tôi, tôi có ADC hoạt động trên SOP8 CH32V003J4M6. Nó sử dụng A0 (PA2) để đọc các giá trị tương tự. Trước đó, có rất nhiều câu hỏi dành cho bạn:
> 
> * Bạn có phiên bản mới nhất của lõi không? (phiên bản 1.0.4)?
> * Bạn đã cập nhật nó với phiên bản cam kết mới nhất của nhánh chính chưa?
> * ADC có được bật không?
> * Bạn đã chọn chưa phù hợp với tốc độ đồng hồ?
> * Bạn đã thử analogRead(A0) thay vì sử dụng PA_2 và bạn chưa sử dụng pinMode(A0, INPUT)?
> 
> Vui lòng đọc thêm số [110](https://github.com/openwch/arduino_core_ch32/issues/110) để biết thêm thông tin về cách sử dụng ADC. Có ví dụ minh họa cách sử dụng ADC đúng cách. Chúc may mắn!

1. I use ARDUINO IDE 2.3.2
2. My code : 
void setup() {
  pinMode(A0, INPUT_ANALOG);
  Serial.begin(9600);
}

void loop() {
  Serial.println(analogRead(A0));
  delay(100);
}

Serial : 
16:02:54.067 -> 664
16:02:54.245 -> 653
16:02:54.473 -> 636
16:02:54.694 -> 670
16:02:54.856 -> 663
16:02:55.081 -> 663
16:02:55.264 -> 665
16:02:55.468 -> 664
16:02:55.672 -> 652
16:02:55.878 -> 631

BUT when i connect pin A0 to the ground or Vcc, this not 
change. 

3. I turn on ADC 
 /* ENABLE Peripherals */
#define                         ADC_MODULE_ENABLED
#define                         UART_MODULE_ENABLED
// #define                         SPI_MODULE_ENABLED
#define                         I2C_MODULE_ENABLED 

My english not good, sorry sir :(

maxint-rd, 2024-09-08:
> My english not good, sorry sir :(

Fortunately we have Google Translate to the rescue!

Your example code looks fine. Just now I flashed it on a J4M6 and for me it works fine. I get zeros when connecting PA2 to GND and 1023 when connecting to 5V. Nice varying values in between when connecting to an NTC thermistor and wild values when disconnected.

**NOTE:** after using serial on the SOP8 chip, you can flash it only once and then you will need to use the WCH-LinkUtility on the LinkE to **_Clear all code flash - by power off_**

I suggest you to check your connections and make sure to copy the latest commits of the master branch over the 1.0.4. files. After enabling ADC you must also empty your build folder to force recompilation of the core code. 
I also suggest you to first try the SOP16 or TSSOP20 module to make debugging easier. With the IDE 2,3.2 you can use real time debugging to step into your code and then go deeper into the core code to see how everything is initialized. 
I'm afraid I don't have any other suggestions for you. I wish you best of luck!

mindonne, 2024-09-08:
:(( 
1. /* ENABLE Peripherals */
// define ADC_MODULE_ENABLED '' turn off ''
#define UART_MODULE_ENABLED
// #define SPI_MODULE_ENABLED
#define I2C_MODULE_ENABLED

this return Zero 

2. 
. /* ENABLE Peripherals */
#define ADC_MODULE_ENABLED '' turn on ''
#define UART_MODULE_ENABLED
// #define SPI_MODULE_ENABLED
#define I2C_MODULE_ENABLED

this return 
17:12:47.140 -> 772
17:12:47.299 -> 746
17:12:47.545 -> 769
17:12:47.732 -> 748
17:12:47.935 -> 776
17:12:48.116 -> 745
17:12:48.344 -> 776

this not change when connecting GND or VCC ( 5v ) 
- board manager 1.0.4 by WCH
- IDE 2.3.2

5V usb to vcc CH32v003j4m6
Gnd to Gnd 
WDIO connect to pin 8 
RX connect to pin 8. 

> I open Wch-Link -> target -> clear all code ... power off 
> I open IDE , my code :
 void setup() {
  pinMode(A0, INPUT_ANALOG);
  Serial.begin(9600);
}

void loop() {
  Serial.println(analogRead(A0));
  delay(100);
} 

ok, up load.
Serial monitor return : 
17:12:47.140 -> 772
17:12:47.299 -> 746
17:12:47.545 -> 769
17:12:47.732 -> 748
17:12:47.935 -> 776
17:12:48.116 -> 745
17:12:48.344 -> 776

I don't know ? why erro :((((

mindonne, 2024-09-08:
Sirrrrrrrrrrr, A0 not working. But use A2 working :V and I think PA2 error

maxint-rd, 2024-09-08:
> Sirrrrrrrrrrr, A0 not working. But use A2 (pin 7) working :V and I think A0 (pin 3) error

That's very interesting. Good to know you have something working.
I've checked my own project connections and I confirm that my project is successfully using A0 (PA2) on pin 3 of the SOP8 CH32V003J4M6. Based on that your chip is giving those values for A0, and on the working other pin, we can conclude that the ADC is working.

In issue #110, you can read about a similar problem. The OP had problems using PA2 because that pin is also used for external crystal. In [this entry](https://github.com/openwch/arduino_core_ch32/issues/110#issuecomment-2237785321) some solution was mentioned. After selecting the proper clock from the clock selection menu (48MHz internal) it should work.

Please check your clock setting and make sure there is no old header file with an override to this setting. If it still doesn't work I suggest you to try a different chip to make sure this one isn't defective and perhaps you should also try the other packages (SOP16/TSSOP20).

mindonne, 2024-09-08:
OMGGGGGGGGGGGGG, I can fix that, I can use A0, A2 on ch32v003j4m6. Thank sirrr

maxint-rd, 2024-09-09:
So you have it working now? Was it the clock setting?

(Btw. issue #94 is another similar case. I guess you're not alone...)

mindonne, 2024-09-10:
> So you have it working now? Was it the clock setting?
> 
> (Btw. issue #94 is another similar case. I guess you're not alone...)

yesss

closed 2025-05-06 by yangliangquan