openwch/arduino_core_ch32 #142: Serial.available() does not work on CH32x035

joraguero, 2024-09-08:
Hi everyone, when I call Serial.available() to see if there is any data in the reception buffer, it always returns -1.

maxint-rd, 2024-09-09:
Hi, this is because it hasn't been implemented (yet).

For my CH32V003 projects I've implemented [interrupt driven Serial1](https://github.com/maxint-rd/arduino_core_ch32/tree/Interrupt-driven-Serial1). Because I only have the v003 I couldn't test any other CH32 chips. Perhaps you want to test this?
My changes add support for available() and peek() and improve read(0 by adding a buffer to avoid loosing bytes when doing other stuff.

To test it shoukd be sufficient to replace these two files:
- cores/arduino/HardwareSerial.h
- cores/arduino/HardwareSerial.cpp

Would be nice if it also works for the CH32x035, but I have not examined that, so no guarantees....

vuhazu, 2024-09-11:
> Hi, this is because it hasn't been implemented (yet).
>
> For my CH32V003 projects I've implemented [interrupt driven [...]

it works for CH32x035, I'm tested

atollminsk, 2024-09-19:
> it works for CH32x035, I'm tested

> Hi, this is because it hasn't been implemented (yet).
>
> For my CH32V003 projects I've implemented [interrupt driven [...]

its work for CH32V208WB. tested.

maxint-rd, 2024-09-19:
@atollminsk  - Excellent. Thank you for testing.

atollminsk, 2024-09-25:
> Hi, this is because it hasn't been implemented (yet).
>
> For my CH32V003 projects I've implemented [interrupt driven [...]

I checked today with CH32V203C8T6 - it works. CH32V307VCT6 - it doesn't work.

maxint-rd, 2024-09-25:
@atollminsk - thank you for your additional testing. Unfortunately I don't have a CH32V307VCT6 so I cannot do further analysis/debugging.

atollminsk, 2024-09-26:
I found the problem, the reason for not working with CH32V307 - I used the command pinMode(x, OUTPUT). With other controllers, the same code worked. I use pinMode(x, OUTPUT_OD).

maxint-rd, 2024-09-26:
@atollminsk - Hmm... So you have tested it working now?
I'm not quite sure what you meant by "Mismatch pin map". Do you mean you used pinMode on the pin you also used for Serial? Or was it an unrelated pin that had an effect on serial communication. (I'm not quite familiar with the pinout of the CH32V307).

atollminsk, 2024-09-26:
@maxint-rd Yes, i used to pinMode pin also for Serial. working with CH32V307)

blopa1961, 2024-10-01:
Serial.available() does not work with the CH32V003 either, but Serial.read() returns a 16 bit -1 value if no byte is pending in the serial port.
So you can use:
mySerialInput= Serial.read();
if (mySerialInput != -1) { etc }

This is a workaround, but I hope this bug is fixed soon!

maxint-rd, 2024-10-01:
> This is a workaround, but I hope this bug is fixed soon!

I already fixed it. See PR #145. That PR also improves Serial.read() to prevent lost data when doing other stuff.(it re-introduces the read-buffer).

blopa1961, 2024-10-01:
> I already fixed it. 

Great!
I tested it when I forced the Board Manager and it didn't work, maybe the old version was catched...
Which file(s) do I need to overwrite to fix Serial.available() ?

atollminsk, 2024-10-02:
@blopa1961 [see this post](https://github.com/openwch/arduino_core_ch32/issues/142#issuecomment-2344029706)

maxint-rd, 2024-10-02:
@blopa1961 - when you open PR #145 you see several tabs. The rightmost lists the files that were changed. In this case [two files changed](https://github.com/openwch/arduino_core_ch32/pull/145/files).
(As mentioned in my first reply of this issue these are HardwareSerial.cpp and its .h)

medtty, 2026-11-07:
Fixed by https://github.com/openwch/arduino_core_ch32/pull/284 ,Serial now receives with an RX interrupt and a 64-byte buffer on every UART and series, so available() returns the number of bytes waiting and nothing is lost while the sketch is busy. Tested on V003, V006, X035, V103, V203 and V307 EVT boards; it will be in the next release

closed 2026-11-07 by medtty