openwch/arduino_core_ch32 #146: SSD1306 for CH32V003j4m6 error !

mindonne, 2024-09-12:
I use library library Adafruit SSD1306 version 2.5.10 for CH32V003j4m6 ,  when compiling : " fatal error: util/delay.h: no such file or directory #include <util/delay.h> "

maxint-rd, 2024-09-13:
Seems like you found that the Adafruit SSD1306 library is not supporting the CH32 core (yet). 
In the [code](https://github.com/adafruit/Adafruit_SSD1306/blob/master/Adafruit_SSD1306.cpp) you can see this:
```
#if !defined(__ARM_ARCH) && !defined(ENERGIA) && !defined(ESP8266) && !defined(ESP32) && !defined(__arc__)
#include <util/delay.h>
#endif
```

The file `util/delay.h` is something specific for the Atmel Arduino core. It provides functions for milli- and microsecond delay. As you can see the Adafruit code doesn't include that file when compiling for various other cores. 

You may get further in compiling this by adding `!defined(CH32V00x)` or `!defined(ARDUINO_ARCH_CH32)` into that `#if`, but you will likely find other issues. I have not yet tried to compile that Adafruit library myself but expect that more modifications are needed. 

What is your project about? Are you just experimenting? 
In a CH32 project for a particular LCD display, I also wanted to ùse Adafruit GFX. The limited flash space of the CH32V003 required me to make a [stripped version](https://github.com/maxint-rd/Adafruit-GFX-CH32V003) of the Adafruit GFX library. Maybe sometime in the future I may find time to attempt to make a CH32 compatible version of Adafruit SSD1306, like I did with [Adafruit_SleepyDog](https://github.com/maxint-rd/Adafruit_SleepyDog), but currently I have other priorities.

BTW: Recently the @hathach added support for [Adafruit qtpy ch32v203 board](#136). In [this post](https://github.com/openwch/arduino_core_ch32/pull/136#issuecomment-2310302748) @ladyada commented that currently no support for other libraries is planned, but that PR's for CH32 support are welcome to be submitted.

mindonne, 2024-09-13:
I just experimenting but library ssd1306 does not support for CH32v003j4m6 :(((

blopa1961, 2024-10-01:
Use the SSD1306ASCII library by Bill Greiman; it works with the CH32V003.

mindonne, 2024-11-25:
> Sử dụng thư viện SSD1306ASCII của Bill Greiman; nó hoạt động với CH32V003.

so good, but .... have to take up more memory :(

maxint-rd, 2024-11-25:
> so good, but .... have to take up more memory :(
I found the CH32X033 to be a nice alternative for projects that don't just fit in the V003.  Although it has some issues, the 64k makes things easier to fit. The TSSOP20 pinout is largely compatible, so I use it as a drop-in replacement during development and debugging, (prior to trying to squeeze code size down to fit in the 16k of the V003).

closed 2026-10-30 by medtty