openwch/arduino_core_ch32 #148: micros() is flawed on CH32X035 (and other targets too probably) overflow management issue

0x0fe, 2024-09-22:
micros() has a seriosu problem on CH32X035, rendring it unusable, this affects any functions that reliles in this function, bascially it contantly overflows or the overflow is not handled correctly, as a result when used to get short timings intervals (less than a millisecond) with a time stamp and substracting the current value to the time stamp a few hunders us before we will often get totally wrong and even negative values. That is, it will randomly return a timestamp inferior to the timestamp taken few hundred microsecond before.
 Obviously that is not how it is supposed to work, micros can overflow after a while, but it should be 70 minutes, not few milliseconds or even less. And even then, the overflow should be handled in the micros function so that it never appears to the function user.

maxint-rd, 2024-09-22:
Could this be related to #65 where @Khalinda found some sort of solution?
Also related to #114?

0x0fe, 2024-09-22:
@maxint-rd possibly, i did not notice it was counting down as in the first issue though. It counts up, except randomly the new value is lower than the previous value (with few hundred us interval).

0x0fe, 2024-09-23:
@maxint-rd 
I tested that fix, from #65 and it seems ok at first glance. 
@TianpeiLee 
What about fixing this issue?

yangliangquan, 2025-04-25:
We have discovered this issue and will correct it in the next version. Thank you！

closed 2025-04-27 by yangliangquan